victronenergy/venus #601: Add support for Smappee energy meters

mansr, 2020-02-04:
Add support for Smappee energy meters. Initially, use only one current sensor per phase, providing functionality similar to other supported meters.

- [ ] dbus-modbus-client changes
- [ ] GUI changes

mpvader, 2020-03-29:
Notes after the review:

- there is a feature now that makes a LED on the CT blink when you have the menu open: would be nice to somehow inform people about that. Now its a bit of a hidden gem. Todo: add a menu line saying: 'To help identification, the LED in this CT is currently blinking'.

To be discussed with Smappee people:
- phantom voltages: we'll need to find out together with Smappee what that means and how to name / document it for our users.
- default configuration would be nice: if there are just three CTs, assume three phase, and assume L1 on the first CT, L2 on the second, L3 on the third. And default to role `Grid`. And if there is just one CT, similarly. To be discussed with Smappee: if you have 3 separate split coil CTs, how does it determine which CT becomes number one, which becomes number two, and so forth.
- Matthijs will call Smappee about above.

mpvader, 2020-07-22:
all done - included in v2.60

closed 2020-07-22 by mpvader