victronenergy/venus #609: Vebus: Device list shows 'Multi' without one being connected

vejpasop, 2020-03-04:
Old feature/bug

Can be reproduced as follows:

1. Connect a Multi
2. Switch it off
3. Remove the Multi and restart Venus

The Multi will be shown in the device list. This is because the Venus device remembered there was a switched off Multi before. 
Has to do with support for old Multi firmware versions (<415) . Without connecting a Multi it is not possible to remove the Multi entry through the user interface.

mpvader, 2023-03-26:
Where is that functionality in venus os @vejpasop ?

vejpasop, 2023-03-31:
This issue can be closed. 
It was fixed by time (probably when we introduced support for Multiple MK3s.)
I tested v2.93 and v3.00~29. In these versions the issue is nonexistent.

closed 2023-03-31 by mpvader