victronenergy/venus #612: VRM Instance / Device instance is confusing

mpvader, 2020-03-17:
In the gui, the Device Instance is meant to be shown, which for canbus products people interpret (ofcourse) as to be the same as the Device Instance the configure on the device.

But, since we recently repurposed /DeviceInstance to be the ` VRM Instance` , its not confusing.

Possible solution: add the real device instance to the D-Bus on a path (or do some renaming all over to remove the confusion), and then show that, or both, in the GUI.

Confirmed by Matthijs on Venus OS v2.60~10 on Thiagos GX Device, but should be easy enough to see elsewhere too.

[image]

[image]

[image]

mpvader, 2020-03-17:
See also here:

https://community.victronenergy.com/questions/44655/venus-os-v2522-available-for-testing.html?childToView=44769#answer-44769

@izak: does this pose problems for DVCC or other things, as also noted by Markus on community? (see above link)

mpvader, 2020-03-18:
As discussed with Jeroen & Izak, this does not pose any issues for DVCC and ESS because the system that drives DVCC (& ESS) control of solar chargers (= vecan-dbus) uses the actual device instance of the solar chargers. Not the number shown in the gui or available on the /DeviceInstance path of each individual solar charger D-Bus service.

The VE.Can Device Instance of (solar) chargers needs to be set to the same Device Instance as the GX Device in order to be part of DVCC and ESS, which Device Instance is by default 0. And at the moment very hard to change: impossible to change from the menus. Also there is normal need to change that away from 0.

mpvader, 2020-06-28:
Latest plan: the VRM instance is going to be called just that: VRM instance, no “device” in the name.

Changes involve only the gui. D-Bus will be left as is.

mpvader, 2020-06-30:
I’ll leave this open for now - but at the moment I dont immediately know what else there is to do in cleaning this up. Anyway, it will become apparent once we release v2.60 and get feedback etc.

jhofstee, 2021-06-04:
Well we can make the dbus more clear about which instance is reported, but that deserves its own issue, so closing this one.

closed 2021-06-04 by jhofstee