victronenergy/venus #613: Phoenix (Smart) Inverters: use better power reading

mpvader, 2020-03-20:
Currently, the apparent power reading as displayed on the GX devices for said products are the result of multiplying current (in 0.1A steps) with voltage. Resulting in a step size of approx 25VA.

Since a while, there is a new register and capability. The task of this issue is to implement using those. VictronConnect already does it properly.

Phoenix Smart Inverters have that better power readout, vreg & capability from the beginning. Our small inverters will soon get an ECN in production adding the same.

The details:

0x0140 VE_REG_CAPABILITIES1 un32 & bit29 = 
Improved load current measurement and support of registers:
VE_REG_AC_OUT_APPARENT_POWER

And two settings, but those are not relevant for GX Devices:
VE_REG_AC_LOAD_SENSE_POWER_THRESHOLD
VE_REG_AC_LOAD_SENSE_POWER_CLEAR

More details of that register in velib, as well as our public PDF: [VE.Direct-HEX-Protocol-Phoenix-Inverter.pdf](https://github.com/victronenergy/venus/files/4358891/VE.Direct-HEX-Protocol-Phoenix-Inverter.pdf), and Hendriek/Cees knows more as well obviously.

Here is how it works now, this is shown when drawing 18W:
[image]

This is shown when drawing 33W:
[image]

ThiemoVanEngelen, 2020-03-23:
vedirect_interface already puts this on the dbus as Ac/Out/L1/S. So checking now what the gui does... Use this path or perhaps it uses Ac/Out/L1/P?

ThiemoVanEngelen, 2020-03-23:
So currently the gui uses Ac/Out/L1/P when available and otherwise /V * /I. The idea is now to prefer /P when available, otherwise /S and as a last resort /V * /I when both are available. When using /P it will show W and otherwise VA.

This will look like this:
[image]

mpvader, 2020-03-31:
Implementation done, next step is for Jeroen to include it.

closed 2020-04-07 by jhofstee