victronenergy/venus #614: dbus-vecan Add VregLink interface

mpvader, 2020-03-24:
This is needed to add remote configuration & firmware updates using VictronConnect for VE.Can-connected products.

Documentation is here in our [dbus wiki document](https://github.com/victronenergy/venus/wiki/dbus-api#the-vreglink-interface)

The bridge between dbus and mqtt is implemented in [mqtt-rpc](https://github.com/victronenergy/mqtt-rpc).

mpvader, 2020-03-24:
Once in place, with regards to VictronConnect itself, this is an over view of support per product:

Products that are immediately supported, since we already support them using Bluetooth or VE.Direct:
- all SmartSolars
- Inverter RS

Products that need work are:
- Lynx Shunt VE.Can (Martin)
- Skylla-IP44 (Roy)
- Skylla-i (Roy)

With regards of priorities for VictronConnect, completing parallel and three phase configuration of inverter/chargers as well as supporting remote Venus OS settings are both (far) more urgent than adding configuration for these products. What would be nice is to at least show them; and then show a popup that remote configuration is not possible. And maybe just show the live data; and no configuration, tbd between Jesus and Matthijs.

Products that we'll not support are:
- The old 150/70 & 150/85 VE.Can BlueSolar Charger (already end of life)

mpvader, 2021-12-15:
moved to v2.90 milestone

martinbosma, 2022-01-11:
Since the VE.Can NACK is implemented in the VE.Can device itself, I noticed that the NACK code may vary per product and comms stack.

VE.Can products have a separate message for the NACK error; additional data is sent in a separate message. 
Where as VE.Direct sends the data and the error value in the same message.

VictronConnect currently supports the following error codes (for the VE.Direct VregLink):
- 0x8101 - Get/Set VREG, unknown register
- 0x8302 - Set VREG, register is read-only
- 0x8304 - Set VREG, invalid or value out of range

For older VE.Can products, i.e. Skylla and Lynx Shunt, in general the following applies:
- 0x8000 - Get VREG, register unknown
  - Report as error code 0x8101
- 0x8100 - Set VREG, register unknown
  - Report as error code 0x8101
- 0x8300 - Set VREG, data written is invalid or out of range (only Lynx Shunt)
  - Drop message and wait for data
- 0x8500 - Message queue overflow 
  - Drop message

For newer VE.Can products that use the lib-vecomm stack, the following applies:
- 0x8001 - Get VREG, register unknown
  - Report as error code 0x8101
- 0x8101 - Set VREG, register unknown
- 0x8102 - Set VREG, register is read-only
  - Report as error code 0x8302
- 0x8104 - Set VREG, register out of range; usually followed by the (clipped) value
  - Drop message and wait for data
- 0x8200 - Busy, try again later; e.g. when out of fast packet buffers
  - Drop message (for now)
- 0x8300 - Invalid; should not occur
  - Drop message
- 0x8500 - Message queue overflow - no need to handle
  - Drop message
- 0x8508 - Message buffer overflow
  - Drop message

closed 2022-04-28 by jhofstee