victronenergy/venus #615: CANVU does not support USB GPSes

SturmUDrang, 2020-03-26:
Tested with a number of devices that work flawlessly on CCGX build of Venus, on CANVU 2.51 they are listed as USB devices but don't get a tty. 

A few examples:

P:  Vendor=1546 ProdID=01a7 Rev=01.00
S:  Manufacturer=u-blox AG - www.u-blox.com
S:  Product=u-blox 7 - GPS/GNSS Receiver

P:  Vendor=1546 ProdID=01a8 Rev=03.01
S:  Manufacturer=u-blox AG - www.u-blox.com
S:  Product=u-blox GNSS receiver

mpvader, 2020-03-26:
thank you for the concise report @SturmUDrang , we'll look into it.

mpvader, 2020-04-15:
Closed in v2.60~somewhere

closed 2020-04-15 by mpvader

mpvader, 2020-04-15:
Hey @SturmUDrang , please try with the Venus OS candidate that we’ll release later tonight. That will have fixed it. Thanks again for reporting