victronenergy/venus #617: Reword unique number test result

mpvader, 2020-03-30:
Add OK. in front. Or something else, the goal is to make it clear that the result means that everything is OK.

[image]

mpvader, 2020-04-02:
@jhofstee I added some more rewording above.

I took `Unique ID Number` from the NMEA2000 document chapter 8.3 Device Information; see screenshot which I took from internet somewhere below.

[image]

vanbergeijk, 2020-04-03:
Please be aware that only the entire NAME is required to be unique. The Unique Identity Number has a dependency on the Manufacturer Code. If other NAME fields are different, e.g. the Function value is different, then it is possible that the same combination of a Manufacturer Code and Unique Identity Number is used by multiple devices. The dependency graph is:

[image]

mpvader, 2020-04-03:
Hi @vanbergeijk , yes absolutely we're aware. Thanks. Changing that setting is only necessary when a users installs multiple GX Devices on one VE.Can network. The issue its solving is quite simply that back when we started producing the GX Devices, we did not program them with a unique number - yet.

jhofstee, 2020-04-04:
@vanbergeijk despite being a nice image, it is unfortunately incorrect for NMEA2000. See 8.3.1.4 of this number across multiple products **of the same function** ... Since this is public information I am tempted to remove above posts to prevent spreading incorrect information. It is correct for J1939 afaik though.

vanbergeijk, 2020-04-07:
@jhofstee, I think we're saying the same: if the function of 2 devices is the same then yes, the manufacturer has to ensure that the identity numbers of those devices are unique. The opposite case of this logic is: when you have 2 devices with different functions, then the same identity number could reoccur in those devices. That's why just an identity number is not enough, the rest of the NAME is required to get to a unique identity.

mpvader, 2020-04-15:
Hi @vanbergeijk and any future readers, FYI all details of how it works with NAME and such in Venus OS are now documented here: https://www.victronenergy.com/live/ccgx:start#nmea2000-out_technical_details

closed 2020-07-12 by jhofstee