victronenergy/venus #618: dbus-fronius in v2.60~ no longer stops scanning when devices are missing

jhofstee, 2020-03-30:
Unlike v2.52, dbus-fronius no longer stops scanning when not all devices are found:

```
2020-03-30 11:57:44.664826500 INFO  IP Address:  QHostAddress( "192.168.4.73" )  Netmask:  QHostAddress( "255.255.252.0" )  
2020-03-30 11:57:44.726756500 INFO  IP Address:  QHostAddress( "169.254.10.65" )  Netmask:  QHostAddress( "255.255.0.0" )  
2020-03-30 11:57:44.804404500 INFO  Registering service "com.victronenergy.fronius" 
2020-03-30 11:57:45.056542500 [VeQItemDbusPublisher] Registered service "com.victronenergy.fronius" 
2020-03-30 11:58:01.277409500 INFO  Not all devices found, starting full IP scan 
2020-03-30 11:58:01.367826500 INFO  IP Address:  QHostAddress( "192.168.4.73" )  Netmask:  QHostAddress( "255.255.252.0" )  
2020-03-30 11:58:01.426589500 INFO  IP Address:  QHostAddress( "169.254.10.65" )  Netmask:  QHostAddress( "255.255.0.0" )  
2020-03-30 12:30:14.789346500 INFO  Auto IP scan completed. Detection finished 
2020-03-30 12:31:00.417927500 INFO  Not all devices found, starting full IP scan 
2020-03-30 12:31:00.421181500 INFO  IP Address:  QHostAddress( "192.168.4.73" )  Netmask:  QHostAddress( "255.255.252.0" )  
2020-03-30 12:31:00.426559500 INFO  IP Address:  QHostAddress( "169.254.10.65" )  Netmask:  QHostAddress( "255.255.0.0" )  
2020-03-30 13:03:08.467697500 INFO  Auto IP scan completed. Detection finished 
2020-03-30 13:04:00.425247500 INFO  Not all devices found, starting full IP scan 
2020-03-30 13:04:00.427851500 INFO  IP Address:  QHostAddress( "192.168.4.73" )  Netmask:  QHostAddress( "255.255.252.0" )  
2020-03-30 13:04:00.428766500 INFO  IP Address:  QHostAddress( "169.254.10.65" )  Netmask:  QHostAddress( "255.255.0.0" )  
2020-03-30 13:36:08.629657500 INFO  Auto IP scan completed. Detection finished 
2020-03-30 13:37:00.510861500 INFO  Not all devices found, starting full IP scan 
2020-03-30 13:37:00.515642500 INFO  IP Address:  QHostAddress( "192.168.4.73" )  Netmask:  QHostAddress( "255.255.252.0" )  
2020-03-30 13:37:00.516538500 INFO  IP Address:  QHostAddress( "169.254.10.65" )  Netmask:  QHostAddress( "255.255.0.0" )
```

In v2.52 it only scans twice.

mpvader, 2020-03-31:
fyi I edited Jeroens comment - adding a todo list.

izak, 2020-04-06:
ARP requests are done by the OS when it needs to translate an IP address to an ethernet address. There is a cache to avoid repeated lookups for often-used addresses, but in general an ARP request (subnet broadcast) will be sent for each IP address. This cannot be changed.

Eliminating the scanning of most of the link-local addresses will remove a million such connection attempts and their accompanying ARP broadcasts.

closed 2020-04-07 by jhofstee