victronenergy/venus #619: Fix custom names for BlueSolarchargers and Smart Solarchargers

mpvader, 2020-03-31:
**Current situation:**

- Our Bluesolar chargers (Roys code stack) doesn't support custom names at all.
- The Smartsolar chargers that don't have VE.Can do, but in a limited manner: the custom name is stored in the BLE chip, and  therefor not available to Venus OS/VRM as well as the Remote feature in VictronConnect.
- The Smartsolar chargers that have VE.Can do have proper custom names support. Also when connected on their VE.Direct port.
- In Venus OS a fallback is active because of all above: you can set a custom name, and thats then stored on the GX Device.

Conclusion: for a Smartsolar charger you can now have two different custom names: one on BLE, and one on Venus OS/VRM.

**Perfect situation:**

One custom name shared all over.

**Roadmap:**

1. Update vedirect-interface to support the custom names (stored in local settings) in the VregLink D-Bus interface. Currently vedirect-interface only supports this fallback in the normal D-Bus interface (ie. the paths interface). This is a small change, can be done by Thiemo, and fixes most confusion when working with VictronConnect remotely.

2. Thiago to fix the Smartsolar chargers to have a custom name & Martijn then to fix the dongle code to support that (or maybe it autodetects this already?). And what would happen with the custom names stored in BLE? Lost? Too bad & acceptable for me, no need to make migration unless really super super simple (`If custom name in ST is default and my own BLE one is not, then update ST custom name end`).

tabreu, 2020-03-31:
About: properly fixing custom names in Roy his MPPT code stack is impossible, due to limited code size

mkuiken, 2020-03-31:
The dongle will prioritize the custom name of the attached SmartSolar (if it has one) over any locally stored custom name. If the name is changed via BLE then the dongle will make a copy and use that copy when the attached SmartSolar indicates that it does not support VE_REG_DESCRIPTION1.

If the name is changed via any other interface on the SmartSolar then there will be a difference with the name stored in the dongle. But as long as the SmartSolar announces its new name to the dongle, then the dongle will prioritize this new name over its locally stored name.

The only situation in which the name stored in the dongle is "lost", is when the SmartSolar was _not_ supporting custom name and then _starts_ supporting it. But as far as I understand from Thiago is this not the case as the VE.Can has been supporting it from the beginning.

tabreu, 2020-03-31:
Good to remember that the SmartSolar VE.Direct also do not support custom names. Only the VE.Can and the new VE.Direct only models will.

jepefe, 2020-03-31:
I noticed that when changing the custom name for example of a VE.Can MPPT from VictronConnect via MQTT, `/CustomName` on DBus doesn't change and Venus has to be restarted to get the new name in the gui. Changing the name via BLE works fine and the name in the Venus GUI is updated immediately.

tabreu, 2020-03-31:
I afraid that is because the PGN 1F016 Configuration Information needs to be sent, when the custom name is changed on mppt, and it is not on the latest FW release. That is a known issue that will be fixed on the next release.

ThiemoVanEngelen, 2020-04-01:
In this case, the MPPT was connected via VE.Direct. What happens is that the messages triggered via mqtt do not follow the normal handling so the answer is only sent back to mqtt not handled internally. This will also be fixed by me.

closed 2020-04-17 by mpvader