victronenergy/venus #620: MFD app not working with Raymarine and Venus >= v2.4x

martinbosma, 2020-04-02:
There has been a report of a user with a Raymarine Axiom MFD where the MFD app is not working properly. The Axiom is able to view the webpage, but shows the error page. The GX device is running v2.52 and both `MQTT on LAN (SSL)` and `MQTT on LAN (Plaintext)` settings are enabled.

When downgrading to v2.42, it still does **not** work and the same error page is shown.

When downgrading to v2.33, it **does** work and the app works as expected.

The Axoim is on the latest firmware ( Lighthouse v3.11), otherwise it would not be able to display the MFD app. Rebooting both devices also does not work.

Error page on Axoim:
[image]

MQTT settings:
[image]

martinbosma, 2020-04-24:
I was able reproduce a similar issue by manually blocking port 9001 on the GX using `firewall deny tcp 9001`. The switch `MQTT on LAN (Plaintext)` remains enable in this case. The GUI normally executes this command, so that would mean that this command has failed. However, this can simply be fixed by toggling the switch to disabled and then enabled again.

This needs to be verified at a customer who suffers from this issue, using `iptables -L -v -n`.

Output of `iptables`
```
Chain INPUT (policy ACCEPT 4288 packets, 496K bytes)
 pkts bytes target     prot opt in     out     source               destination         
72968   12M ACCEPT     all  --  lo     *       0.0.0.0/0            0.0.0.0/0           
17885 2420K new-conn   all  --  *      *       0.0.0.0/0            0.0.0.0/0            ctstate NEW

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain OUTPUT (policy ACCEPT 18125 packets, 1998K bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain new-conn (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    2   128 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:22
    0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:8883
    0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:1883
    0     0 REJECT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:9001 reject-with icmp-port-unreachable

```

mpvader, 2020-05-19:
[Here](https://github.com/victronenergy/venus/issues/378#issuecomment-630348799) a user reports that his local LAN MQTT didn't work until he resetted a token on VRM. Perhaps Mosquitto doesn't work (well) anymore if it has an issue in sending data to VRM because of wrong password.

I'm on the phone with @wiebeytec now and he'll test above - simple enough to test. I'm putting it here in this issue since it might be related.

martinbosma, 2020-05-19:
I've seen this issue also with newly installed GX-es and even without internet connection. Can that still be related?

mpvader, 2020-05-19:
> I've seen this issue also with newly installed GX-es and even without internet connection. Can that still be related?

I don't know. But its leading to the same, -sofar unexplained- fault. Lets just see what Wiebe finds out wrt how the bridge config affects Mosquittos stability on local LAN queries.

wiebeytec, 2020-05-19:
See branch. I think I fixed it.

The bridge config is (now?) a symlink (`/run/mosquitto/vrm_bridge.conf`), and was dead by default, until a bridge config was successfully written (which requires network). Mosquitto crashed on it.

dbus-mqtt and mqtt-rpc need the patched `velib_python`.

Side note: the mosquitto logs get spammed with bridge connection errors if you enable that and have no network, or the auth fails.

martinbosma, 2020-05-19:
I was able to reproduce by removing `/data/conf/mosquitto.d/vrm_bridge.conf` and restart mosquitto and block internet and fixing it by running `touch /data/conf/mosquitto.d/vrm_bridge.conf` and restart mosquitto.

wiebeytec, 2020-05-20:
To answer some questions / add some clarification:

* New installs of the current Venus 2.40 through 2.53 will _not_ have functional LAN MQTT until there is an internet connection. No setting can work around it.
* Same goes for reflashed Raspberry's (and others? Those that lose their data partition) until there is an internet connection, _and_ a token reset is performed on VRM.
* Any setting applies: VRM two way communication, MQTT LAN. Mosquitto will just not run.
* Commit 6b69825 (2019-10-16 13:28) in meta-victronenergy introduced it, where the Mosquitto config included a dir from `/run`, namely `/run/mosquitto`, which can contain a dead link. This was released first in Venus 2.40.
* Upgrading to a future version with my fix will fix it, no special action required.
* The user can fix it by logging in with SSH and running `touch /data/conf/mosquitto.d/vrm_bridge.conf`

wiebeytec, 2020-05-20:
Fix merged in master of mqtt-rpc and dbus-mqtt and put on the TODO for Venus 2.60.

mpvader, 2020-05-21:
closed after adding the "fix path creation bug" commit. Fixes are in v2.60~23, and also backported to v2.54.

Another issue was uncovered and still open, see #642

closed 2020-05-21 by mpvader