victronenergy/venus #623: Cerbo GX Touch issues

mpvader, 2020-04-15:
There are two similar yet separate issues reported:

### 1. Touch (& mouse press) events stop working

as reported by Guy:
> When changing the relay settings back and forth several times from Generator to Tank pump,
> The screen become unresponsive in the settings menu.
> Wasn't totally frozen, the 'demo' animation still appeared (flashing), but would not accept input from the touch screen.
> I could still remotely control normally it via remote console on the local network (keyboard, didn't try the mouse), but it would no longer accept any touch on the GX Touch.
> A soft reboot (via remote console, General menu) and the touch screen worked again. I was unable to reproduce it, after rebooting.

And also seen by Lucian; and then looked into by Mans & Jeroen. Sofar the conclusion is that (a) the screen blanking & waking it up still works. The mouse press events do make it to the gui. Something (probably) goes wrong in QDeclarativeView::mousePressEvent.

### 2. GX Touch display freezes, even after power cycling

As reported by Lucian:

> I have some feedback (for Cerbo), sometimes the screen is freezing. The remote console is working but not on the lcd. Do you have other reports? I heard it from 2 clients. I also saw it but on mine, having hardware problems, I was not sure to report this or not. Only solution is to reboot it
> Even though operating it from remote console worked well the GX Touch kept displaying the old 'image'.
> the customers were describing the same behaviour.
> disconnecting hdmi and usb didn't help: once reconnected the GX Touch powered up again and and showed the same old image again.
> I'm sure the screen was not updating, not sure if the touch was still working or not.

mpvader, 2020-04-15:
For now, we've been unable to reproduce it. Next step is adding some logging, so next time someone reports it we have more data.

Wholistic, 2020-04-16:
This is vague enough to be confusing, and could be completely unrelated, but I don't know where else to put it. 

I have also experienced another 'display frozen' situation with the MultiPlus-II GX, where the LCD screen is frozen, button does nothing, and values do not update. But it is still perfectly good via remote console, and for power electric functions. 

https://www.dropbox.com/s/l86bz8oicl7ae5s/multiplus-II%20gx%20frozen.mp4?dl=0

mpvader, 2020-04-16:
Yes unrelated. I’ll make a separate issue later today.

mpvader, 2020-06-30:
Probably both issues are solved now - if there even were two to begin with. Closing.

closed 2020-06-30 by mpvader