victronenergy/venus #624: Add support for new USB analog input device

mpvader, 2020-04-15:
Its a USB to I2C converter + I2C-ADC.

Update kernel configuration cq enable drivers, and test it with dbus-adc

mpvader, 2020-04-16:
Summary of what was discussed in Slack:

In the Victron world, there are three types of tank level senders, 0-10V Voltage, resistance and 4 to 20mA/
This new usb-adc device is to add 0-10V and 4 to 20mA capability to our product family. Selecting between them is done with a (software) switch. We could change that to hardware if necessary to simplify things: for the installer that makes no difference: setting it up in software versus setting a dip-switch.
For the 4 to 20mA type, the hardware cannot power the sender. (some other 4-20mA input type devices can, which is why I'm noting it here)

Reference for the sender aka transducer used in the design is here: https://www.nieuwkoopbv.nl/wp-content/uploads/2015/06/NT5000-Nivosensor-KTE8000-Eng.pdf

I've asked for more feedback on Community:
https://community.victronenergy.com/questions/47324/looking-for-input-4-to-20ma-senders.html

Felix added:
> there are 4 gpios which have to be controlled, (i think the default state is current if not controlled, if i remember correctly). that gpio functionality is already in the driver available, it reports itself als usb->i2c bridge and also as gpio interface. i controlled the gpios once to test that.

mpvader, 2020-04-17:
First conclusion from the feedback requested on community: min and max levels need to be configurable. And lets make it complete then, same for voltage. @mansr

mkrenzke, 2020-04-20:
Also Community mentioned to calibrate it to more than just two points (full & empty). This obviously makes sense for odd-shaped tanks which are quite common in vehicles. But this makes it way more complicated I'd assume.

mpvader, 2020-06-05:
New prototypes were ordered and expected to be ready this week or next. Schematic of the new revision (A2) here: https://github.com/victronenergy/venus-hardware/tree/master/gx_tanksens/USB_Tanksensor_A2

Here is a picture showing the device:

[image]

Specifications:
- A total of 4 analog input channels
- All channels are configurable in software to measure either 0-10V or 4-20mA
- Channel 1 and 2 include a built-in 24V power supply
- Channel 3 and 4 do not. But there is an extra connector to facilitate easy wiring
- All channels feature an internal 20mA self-resetting fuse
- All channels are protected: accidentally selecting the wrong input type will not break the analog input module nor the sensor.

Channel labelling:
- Vout    - Power supply to sensor
- In      - Analog input. For current sensors this is the return path, and for voltage sensors this is the analog input.
- GND     - Ground. Use for Voltage sensors only.

mpvader, 2021-02-19:
## Test results

### migration
Original version: v2.63 + fix by Izak to sort the xml by tag-name recursively.
Test version 20210208211007-v2.70 by Mans + fix same fix by Izak again.

Various settings made in builtin tank inputs 1, 2 and 4 on a Cerbo:
- capacity (201, 202 and 204l)
- measurement type (eur, us, custom 4 & 184Ohm)
- tank type (fresh, waste, oil)
- custom shape (10/1 + 20/20, 1/1 + 2/2, 50/50 + 90/90)
- custom names (T1, TL2, TL4)

And tank input 3 was disabled.

Various settings made for the temp inputs:
- type (fridge, generic, fridge)
- custom name (TS1, TS2, TS4)

Then updated to above mentioned v2.70 version.

Then checked that all settings were migrated:
- settigns.xml version increased to 9 OK.
- all above settings including their custom names and enabled / not-enabled state were all migrated ok.

See below the diff of settings.xml before and after. A patch by Izak was used to make localsettings store the xml sorted by tag name.

### rest of tests

- functionality of the new filtering feature; works good. Set it to 1s and result is immediate. Set it to its max, 60s, and it takes 60 seconds to creep to the new value.
- unplugging / plugging of the GX tank. Works fine. dbus-adc gets restarted, which gives a brief bleep on the gui on the built-in tank inputs (data goes away and comes back) but thats expected and no problem.
- gx tank measurements are on by default; built-in tanks are still off by default: OK.
- when the type is set to 4-20mA, shorting the input gives an error: OK.
- naming: setting a custom name works.
- the default naming is a bit confusing, ie you don't immediately know which is input 1, which 2, which 3 and which is 4. See also screenshots below), but if I recall correctly there is no very simple better way - and regardless: no need to further delay market introduction for this. Which input is what is easily seen in the Device menu (see screenshot).
- numbering is OK: left input is input 1, right input is GX Tank input 4.

And here is what dmesg says on plugin:
```
[  357.320599] usb 4-1: new full-speed USB device number 4 using ohci-platform
[  357.593649] usb 4-1: New USB device found, idVendor=10c4, idProduct=ea90, bcdDevice= 0.00
[  357.601958] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  357.609120] usb 4-1: Product: GX Tank 140
[  357.613213] usb 4-1: Manufacturer: Victron Energy BV
[  357.618205] usb 4-1: SerialNumber: HQ2051JVWW6
[  357.631903] cp2112 0003:10C4:EA90.0003: hidraw0: USB HID v1.01 Device [Victron Energy BV GX Tank 140] on usb-1c1c400.usb-1/input0
[  357.711658] cp2112 0003:10C4:EA90.0003: Part Number: 0x0C Device Version: 0x03
[  357.746854] at24 5-0050: 256 byte 24c02 EEPROM, writable, 1 bytes/write
[  357.753649] i2c i2c-5: new_device: Instantiated device 24c02 at 0x50
[  357.761251] i2c i2c-5: new_device: Instantiated device tla2024 at 0x48
```

and on removal:
```
[  386.029691] usb 4-1: USB disconnect, device number 4
```

Picture 1: default look when inserted. fuel tank (28) is input 4 on the GX Tank, fuel tank (31) is input 1.

[image]

Picture 2: device page of a GX Tank item

[image]

Picture 3: Enabling/disabling the inputs:

[image]

### migration diff:
[... 214 lines cut ...]
-    <_1>
-      <Offset type="f" min="-100.0" max="100.0" default="0.0" silent="False">0.0</Offset>
-      <Scale type="f" min="0.10000000149" max="10.0" default="1.0" silent="False">1.0</Scale>
-      <TemperatureType2 type="i" min="0" max="2147483644" default="0" silent="False">1</TemperatureType2>
-    </_1>
-    <_2>
-      <Offset type="f" min="-100.0" max="100.0" default="0.0" silent="False">0.0</Offset>
-      <Scale type="f" min="0.10000000149" max="10.0" default="1.0" silent="False">1.0</Scale>
-      <TemperatureType2 type="i" min="0" max="2147483644" default="0" silent="False">2</TemperatureType2>
-    </_2>
-    <_3>
-      <Offset type="f" min="-100.0" max="100.0" default="0.0" silent="False">0.0</Offset>
-      <Scale type="f" min="0.10000000149" max="10.0" default="1.0" silent="False">1.0</Scale>
-      <TemperatureType2 type="i" min="0" max="2147483644" default="0" silent="False">0</TemperatureType2>
-    </_3>
-    <_4>
-      <Offset type="f" min="-100.0" max="100.0" default="0.0" silent="False">0.0</Offset>
-      <Scale type="f" min="0.10000000149" max="10.0" default="1.0" silent="False">1.0</Scale>
-      <TemperatureType2 type="i" min="0" max="2147483644" default="0" silent="False">1</TemperatureType2>
-    </_4>
-  </Temperature>
   <Vecan>
     <can0>
       <FreeIdentityNumber type="i" min="0" max="500" default="0" silent="False">1</FreeIdentityNumber>
```

mpvader, 2021-02-21:
Updated above a little bit, after realising I had only tested leaving the gx tank input open when set to mA-s. Not testing what happens when shorting it

mpvader, 2021-03-10:
in master / v2.70 as well as v2.65. closing.

closed 2021-03-10 by mpvader