victronenergy/venus #627: Issue with EasySolar character display?

mpvader, 2020-04-16:
As reported by Guy:

> I have experienced a 'display frozen' situation with the MultiPlus-II GX, where the LCD screen is frozen, button does nothing, and values do not update. But it is still perfectly good via remote console, and for power electric functions. 
> https://www.dropbox.com/s/l86bz8oicl7ae5s/multiplus-II%20gx%20frozen.mp4?dl=0

Next step: when it happens again, Guy will leave it as is and contact us to login and see whats wrong.

Wholistic, 2020-10-13:
Not me, but here is another active case:

marek@apn.pl

https://vrm.victronenergy.com/installation/71507/overview

https://community.victronenergy.com/questions/63693/gx-display-on-my-easysolar-ii-gx-is-frozen-again.html

Wholistic, 2020-10-20:
User reported back, and unit is currently in the frozen state for us to investigate

izak, 2020-10-21:
Alright. There is one bug (not the real one, but one preventing finding the real one). We need to start dbus-characterdisplay with stdout unbuffered (python -u), otherwise we'll never see the exception logged. I changed this in place at this installation. Are you in contact with the user @Wholistic or should I contact him?

izak, 2020-10-22:
Some feedback: Reloading the HD44780 driver makes it recover. This is either a hardware issue or a kernel driver issue.

Steps to reload/rebind driver (because it is compiled into the kernel):

```
root@nanopi:~# echo display >/sys/bus/platform/drivers/hd44780/unbind 
root@nanopi:~# echo display >/sys/bus/platform/drivers/hd44780/bind
```

jhofstee, 2021-06-04:
I will remove the milestone from this issue, since it is around for more then a year now. It seems to be unknown how to reproduce this (and hence how to fix it).

mpvader, 2021-06-28:
Impossible to reproduce, Mans tried, perhaps others as well; and no complaints anymore. Closing. If the problem is still out there, it will come back.

closed 2021-06-28 by mpvader

RobDuthie, 2021-06-28:
Hi Yes i have had a easysolar with frozen screen in tha last month for no apparant reason, Just had to reset unit to get reading again.
Regards
Rob D
NZ