victronenergy/venus #631: GPS afterwork

mpvader, 2020-04-26:
Since we now support reading gps data from NMEA2000, there can be multiple GPS-es on the dbus. Because of that, our serial gps driver, dbus-gps (or gps-dbus) used to publish on com.victronenergy.gps, is now updated to now have a fourth segment in its dbus service name. And it publishes a /DeviceInstance path. That /DeviceInstance path is a VrmInstance: a number reserved through local settings starting at 0.

VRM, for its location & geofence services, needs to know one set of GPS data, which it expects at instance 0.

The gui also needs to know one set of GPS data.

The plan:

- [ ] Update systemcalc to publish one set of gps data under com.victronenergy.system. Out of the list of available GPS-es with a fix, take the one with the lowest /DeviceInstance.
- [ ] Update vrmlogger to take the data from com.victronenergy.system, rather than com.victronenergy.gps  (there is no need to push all GPS-es to VRM)
- [ ] Restore the GUI status bar functionality related to gps (mobile overview & status bar); see [this commit](https://github.com/victronenergy/gui/commit/93c1b7099f1b7a21fcba0de1db1ebb25c8221d1f).

mpvader, 2020-04-28:
I updated above to the new plan

jhofstee, 2020-05-11:
include in v2.60~19, https://github.com/victronenergy/gui/commit/129df42a96cebae9e220b58dd5319302c34e4732

closed 2020-05-11 by jhofstee