victronenergy/venus #632: USB modem iPhone

hanschristiangithub, 2020-04-26:
I am trying to use an iPhone as an USB modem/gps on pi 3+ with Venus os 2.42. iPhone USB modem is recognized and worked when connected with pc/Mac. When connected to pi iPhone charges via usb but is not recognized as gsm modem nor as gps device.

mpvader, 2020-04-26:
Hi @hanschristiangithub, yes indeed that doesnt work - sorry.

I’ll close this issue, its not something we’re planning to make work.

closed 2020-04-26 by mpvader

hanschristiangithub, 2020-04-26:
Hi @mpvader, thank you for the quick response. I also have a quectel 25-A card with a sixfab hat that is not being recognized. Is in this case a solution available? Thank you for your help.

mpvader, 2020-04-27:
Hi, no not either. The only working modem, as far as I know, is our own GX GSM. See manual.

Others might be able to make working, but then you’ll need to go in and change the source code