victronenergy/venus #642: mqtt issue in order of booting & connecting internet

mpvader, 2020-05-21:
After the wild fix of https://github.com/victronenergy/venus/issues/620, there is still a remaining issue., as found by Jeroen.

To reproduce: Take a new device, ie one that doesn't yet has performed successfull vrm bridge configuration. Boot it up while it doesn't have internet. And then connect internet only later.

Expected behaviour: Once there is internet, dbus-mqtt or mqtt-rpc will successfully do the registration at VRM (since they will retry forever after first failing because there is no internet), and Mosquitto recognises that and connects to its online MQTT broker.

Current behaviour: registration works, but mosquitto doesn't recognise that there is an updated bridge-config file.

This might be because the file is there as a symlink, but I did a quick test by modifying one of the ports in /etc/mosquitto/mosquitto.conf, and mosquitto did not pick that up.. At least not within a few seconds. I'll keep it running to see if checks those config files on an interval.

mpvader, 2020-05-21:
Note that the velib python mosquitto bridge registrator python also has code to restart the broker:

https://github.com/victronenergy/velib_python/blob/master/mosquitto_bridge_registrator.py#L216

mpvader, 2020-05-21:
And also some twenty minutes later: mosquitto didnt by itself pick up the port nr change I made to /etc/mosquitto/mosquitto.conf. Restarting mosquitto does make it pick it up

wiebeytec, 2020-05-22:
You didn't say what MQTT services you enabled on the 'new device'. I suspect only vrm-two-way communcation. As known, the bridge registrator is shared between dbus-mqtt and mqtt-rpc, but while dbus-mqtt [starts it with the default option to restart the bridge on submission of token](https://github.com/victronenergy/dbus-mqtt/blob/647fc232c476eb0318bb041efac817a8756f4ce6/dbus_mqtt.py#L59), the [mqtt-rpc explicitly says restart=false](https://github.com/victronenergy/mqtt-rpc/blob/daea938dc5ab50d75370c01e3e4562a6282b2215/mqtt-rpc.py#L806).

wiebeytec, 2020-05-26:
See the associated commit in a branch in velib_python. I tested it by:

* Turning off all mqtt related settings
* removing `/data/conf/mosquitto.d/` (I also did runs with having an empty `/data/conf/mosquitto.d/vrm_bridge.conf` as well)
* Blocking outgoing connections to the registration server before enabling vrm-two-way comm.
* Both with and without also deleting `/data/conf/mqtt_password.txt`

If agreed, I can release a dbus-mqtt and mqtt-rpc with this fix.

mpvader, 2020-05-27:
Agreed @wiebeytec - pls complete the fix

wiebeytec, 2020-05-27:
Fixes released, and put them on the Venus TODO.

jhofstee, 2020-06-05:
included in v2.60~24

closed 2020-06-05 by jhofstee