victronenergy/venus #652: dbus-vecan crash

jhofstee, 2020-06-16:
2020-06-15 03:21:56.250475500 -------- dumping backlog -------
2020-06-15 03:21:56.250482500 936.812 INF.dbus_forward: freeing dbus forward service com.victronenergy.solarcharger.ttyO2
2020-06-15 03:21:56.250491500 949.939 INF.: device disconnected: nad 0x24 1523
2020-06-15 03:21:56.250498500 615.828 INF.dbus_forward: fowarded dbus service com.victronenergy.solarcharger.ttyO2
2020-06-15 03:21:56.250506500 615.940 INF.senders: got nad 36 for solarcharger
2020-06-15 03:21:56.250513500 615.950 INF.senders: got instance 0 for solarcharger
2020-06-15 03:21:56.250891500 615.956 INF.senders: got system instance 0 for solarcharger
2020-06-15 03:21:56.250901500 616.063 INF.senders: got identity number 1523 for solarcharger
2020-06-15 03:21:56.250909500 616.072 INF.senders: start ACL for solarcharger
2020-06-15 03:21:56.250915500 007.206 INF.dbus_forward: freeing dbus forward service com.victronenergy.solarcharger.ttyO2
2020-06-15 03:21:56.250924500 019.664 INF.: device disconnected: nad 0x24 1523
2020-06-15 03:21:56.251032500 210.904 INF.dbus_forward: fowarded dbus service com.victronenergy.solarcharger.ttyO2
2020-06-15 03:21:56.251042500 211.034 INF.senders: got nad 36 for solarcharger
2020-06-15 03:21:56.251049500 211.039 INF.senders: got instance 0 for solarcharger
2020-06-15 03:21:56.251056500 211.067 INF.senders: got system instance 0 for solarcharger
2020-06-15 03:21:56.251063500 211.203 INF.senders: got identity number 1523 for solarcharger
2020-06-15 03:21:56.251100500 211.205 INF.senders: start ACL for solarcharger
2020-06-15 03:21:56.251181500 246.236 INF.dbus_forward: freeing dbus forward service com.victronenergy.solarcharger.ttyO2
2020-06-15 03:21:56.251191500 250.350 INF.dbus_forward: fowarded dbus service com.victronenergy.solarcharger.ttyO2
2020-06-15 03:21:56.251199500 250.426 INF.senders: got nad 36 for solarcharger
2020-06-15 03:21:56.251205500 250.432 INF.senders: got instance 0 for solarcharger
2020-06-15 03:21:56.251212500 250.439 INF.senders: got system instance 0 for solarcharger
2020-06-15 03:21:56.251246500 250.531 INF.senders: got identity number 1523 for solarcharger
2020-06-15 03:21:56.251255500 250.538 INF.senders: start ACL for solarcharger
2020-06-15 03:21:56.251261500 490.890 INF.dbus_forward: freeing dbus forward service com.victronenergy.solarcharger.ttyO2
2020-06-15 03:21:56.251270500 503.900 INF.: device disconnected: nad 0x24 1523
2020-06-15 03:21:56.251276500 120.399 ERR.Variant: veVariantInvalidate: Unknown type!
2020-06-15 03:21:56.259543500 process 18370: arguments to dbus_message_new_method_call() were incorrect, assertion "_dbus_check_is_valid_path (path)" failed in file ../../dbus-1.10.20/dbus/dbus-message.c line 1371.

jhofstee, 2020-06-16:
Jun 15 03:21:56 ccgx user.info kernel: [315998.115091] potentially unexpected fatal signal 6.
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115164] CPU: 0 PID: 18370 Comm: vecan-dbus Not tainted 4.19.74 #1
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115187] Hardware name: Generic AM3517 (Flattened Device Tree)
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115219] PC is at 0xb6cd69fc
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115240] LR is at 0xbefc26c0
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115262] pc : [<b6cd69fc>]    lr : [<befc26c0>]    psr: 000d0010
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115283] sp : befc26c0  ip : 00000000  fp : 00000000
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115304] r10: 0001a0d4  r9 : b6f0f000  r8 : befc2950
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115327] r7 : 000000af  r6 : 00033b22  r5 : b6de50d8  r4 : 00000006
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115348] r3 : 00000008  r2 : 00000000  r1 : befc26c0  r0 : 00000000
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115374] Flags: nzcv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115396] Control: 10c5387d  Table: 8533c019  DAC: 00000055
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115420] CPU: 0 PID: 18370 Comm: vecan-dbus Not tainted 4.19.74 #1
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115440] Hardware name: Generic AM3517 (Flattened Device Tree)
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115503] [<c010e3a0>] (unwind_backtrace) from [<c010c558>] (show_stack+0x10/0x14)
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115541] [<c010c558>] (show_stack) from [<c0138338>] (get_signal+0x830/0x8ac)
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115573] [<c0138338>] (get_signal) from [<c010b974>] (do_signal+0x1ec/0x4c8)
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115604] [<c010b974>] (do_signal) from [<c010be30>] (do_work_pending+0xd0/0xf4)
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115637] [<c010be30>] (do_work_pending) from [<c010106c>] (slow_work_pending+0xc/0x20)
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115661] Exception stack(0xc50d9fb0 to 0xc50d9ff8)
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115689] 9fa0:                                     00000000 befc26c0 00000000 00000008
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115719] 9fc0: 00000006 b6de50d8 00033b22 000000af befc2950 b6f0f000 0001a0d4 00000000
Jun 15 03:21:56 ccgx user.warn kernel: [315998.115746] 9fe0: 00000000 befc26c0 befc26c0 b6cd69fc 000d0010 00000002

mpvader, 2020-06-19:
Relatively harmless: this happens while shutting down.

closed 2020-07-12 by jhofstee