victronenergy/venus #654: Venus GX - failed emmc? or more?

mpvader, 2020-06-21:
An returned unit of which the complaint was that it no longer worked.

Output on the serial console:

```
[    1.810530] VFS: Mounted root (ext4 filesystem) on device 179:10.
[    1.817845] devtmpfs: mounted
[    1.822325] Freeing unused kernel memory: 1024K
[    1.830634] EXT4-fs warning (device mmcblk1p2): dx_probe:765: inode #8822: co                                                                                                                                                                             mm swapper: Unrecognised inode hash code 47
[    1.842021] EXT4-fs warning (device mmcblk1p2): dx_probe:870: inode #8822: co                                                                                                                                                                             mm swapper: Corrupt directory, running e2fsck is recommended
[    1.857137] Kernel panic - not syncing: No working init found.  Try passing i                                                                                                                                                                             nit= option to kernel. See Linux Documentation/init.txt for guidance.
[    1.870858] ---[ end Kernel panic - not syncing: No working init found.  Try                                                                                                                                                                              passing init= option to kernel. See Linux Documentation/init.txt for guidance.
```

version in uboot was 1. Changed to 2. Power cycled.

Bootup now worked ok. installed version v2.53. But before I was able to do anything else this happened:

```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.194-venus (oe-user@oe-host) (gcc version 7.3.0 (GCC) ) #1 Sun Apr 5 14:37:11 UTC 2020
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Venus GX
[    0.000000] cma: Reserved 16 MiB at 0x9f000000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyO0,115200n8 consoleblank=0 root=/dev/mmcblk1p3 rw rootfstype=ext4 rootwait 5 omap_wdt.early_enable=1
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 494380K/524288K available (5120K kernel code, 298K rwdata, 1464K rodata, 1024K init, 276K bss, 13524K reserved, 16384K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0600000   (6112 kB)
[    0.000000]       .init : 0xc0800000 - 0xc0900000   (1024 kB)
[    0.000000]       .data : 0xc0900000 - 0xc094ab50   ( 299 kB)
[    0.000000]        .bss : 0xc094c000 - 0xc0991208   ( 277 kB)
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000014] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000031] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000039] OMAP clocksource: timer1 at 24000000 Hz
[    0.000333] clocksource_probe: no matching clocksources found
[    0.000555] Console: colour dummy device 80x30
[    0.000578] Calibrating delay loop... 996.14 BogoMIPS (lpj=4980736)
[    0.089352] pid_max: default: 32768 minimum: 301
[    0.089463] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.089473] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.089988] CPU: Testing write buffer coherency: ok
[    0.090036] CPU0: Spectre v2: using BPIALL workaround
[    0.090280] Setting up static identity map for 0x80100000 - 0x80100058
[    0.091985] devtmpfs: initialized
[    0.100388] VFP support v0.3: implementor 41 architecture 3 part 30 variant c rev 3
[    0.100656] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.100677] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.101681] pinctrl core: initialized pinctrl subsystem
[    0.102575] NET: Registered protocol family 16
[    0.104136] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.114777] omap_hwmod: debugss: _wait_target_disable failed
[... 727 lines cut ...]
Jun 21 12:08:53 beaglebone user.err kernel: [   51.790425] mmcblk1: error -110 sending status command, aborting
Jun 21 12:08:53 beaglebone user[   51.818264] mmcblk1: error -110 sending status command, retrying
.err kernel: [   51.801194] mmcblk1: error -110 sending status command, retrying
Jun 21 12:08:53 beaglebone user.err kernel: [   51.818264] mmcblk1: error -110 sending status command, retrying
[   51.853320] mmcblk1: error -110 sending status command, aborting
Jun 21 12:08:53 beaglebone user.err kernel: [   51.853320] mmcblk1: error -110 sending status co[   51.870279] mmcblk1: error -110 sending status command, retrying
mmand, aborting
[   51.890763] mmcblk1: error -110 sending status command, retrying error -110 sending status command, retrying

Jun 21 12:08:53 beaglebone user.err kernel: [   51.890763] mmcblk1: error -110 sending status command, retrying
[   51.913102] mmcblk1: error -110 sending status command, aborting
Jun 21 12:08:53 beaglebone user.err kernel: [   51.913102] mmcblk1: error -110 sending status command, aborting
[   51.932601] mmcblk1: error -110 sending status command, retrying
[   51.950711] mmcblk1: error -110 sending status command, retrying error -110 sending status command, retrying

Jun 21 12:08:53 beaglebone user.err kernel: [   51.950711] mmcblk1: error -110 sending status command, retrying
[   51.975862] mmcblk1: error -110 sending status command, aborting
[   51.992014] mmcblk1: error -110 sending status command, retrying error -110 sending status command, aborting

Jun 21 12:08:53 beaglebone user.err kernel: [   51.992014] mmcblk1: error -110 sending status command, retrying
[   52.018365] mmcblk1: error -110 sending status command, retrying
Jun 21 12:08:53 beaglebone user.err kernel: [   52.018365] mmcbl[   52.036506] mmcblk1: error -110 sending status command, aborting
```

And then I pulled the plug to make sure the initial part of log output wasn't lost due to buffer size.

aaronsb, 2020-06-22:
This is nearly identical to the type of behavior I've seen on a raspberry
pi with an emmc card that I've (ab)used before. Now days any raspberry pi I
use has a high life card like a samsung pro endurance card, or I started
using the sd card as a boot stub to a usb mounted SSD drive.

I don't know the particulars about how the venus os is partitioned or
leverages disk as temporary location, but minimizing writes is #1 to
prevent emmc failure. (I'm sure you know this!)

I don't think venus os makes use of F2FS does it?
https://en.wikipedia.org/wiki/F2FS

On Sun, Jun 21, 2020 at 5:11 AM mpvader <notifications@github.com> wrote:

> An returned unit of which the complaint was that it no longer worked.
>
> Output on the serial console:
>
> [    1.810530] VFS: Mounted root (ext4 filesystem) on device 179:10.
> [    1.817845] devtmpfs: [...]

jhofstee, 2026-09-05:
@aaronsb no, we don't have F2FS. It is ubifs and ext4 at the moment. The rootfs is read-only though.
Anyway, as this is six year old, and never commented on more, closing as an incident.

closed 2026-09-05 by jhofstee