victronenergy/venus #661: Add VE.Bus "High DC Voltage alarm"

mpvader, 2020-06-29:
- [ ] Transmit it (Peter B)
- [ ] Get it on the dbus (Jos)
- [ ] Show it in the gui and make a notification as well
- [ ] Add to VRM Logger

That this is missing is confusing, since other than this alarm, you see nothing.

And it can happen, especially in case a managed battery disconnects it self.

For example VRM site 63025, at 10:58 in the morning Europe time.

vejpasop, 2020-07-02:
All VE.Bus devices will communicate their own "High DC voltage alarm" flag.
I propose to combine (OR) these in a single "/Alarms/HighDCVoltage" D-Bus path. 

"Devices/#/ExtendedStatus/HighDCVoltage" paths will be added to show the individual per device status.

The Alarm will indicate why the system switched off,and the extended status can be used to pinpoint the cause.

mpvader, 2020-07-02:
yes sounds good to me

vejpasop, 2020-07-07:
Along with HighDCVoltage the Multi firmware >=475 now also reports HighDCCurrent which currently is another condition causing the Multi to switch off without any status.  

I will add "Alarms/HighDCCurrent" also

mpvader, 2020-07-07:
And in what conditions can that occur? (High dc current). And How long must it endure on the multi Before its raised?

vejpasop, 2020-07-08:
When the DC voltage is too low for the transformer to reach its output voltage, the current flow will be too high. As a protection the Multi will switch off and stay off. This could happen when for instance in a 12V battery a cell is shorted and the battery voltage drops below 8V.

mpvader, 2020-07-27:
Hey @vejpasop , I see below  paths on D-Bus on Groningen Hub-1, but thats running an older firmware version. So two questions:

1) What are the exact new paths that this change adds, including additionally added paths.
2) Doesnt mk2-dbus normally hide paths in case they concern features not supported by the system? I think it does and should do that also for these. Otherwise, for example on VRM, its very confusing for the user.

closed 2026-10-07 by mpvader