victronenergy/venus #662: VE.Bus: Ignore AC input status does not update when swiching from AC in 2 to  back to 1 on Quattro

vejpasop, 2020-06-29:
```/Ac/State/IgnoreAcInput``` is supported since mk2-dbus v3.04, Venus v2.60~17

**Bug description:**
1) Enable "Ignore AC input" on AC in 1 (use general flag assistant for this)
2) ```/Ac/State/IgnoreAcInput``` updates to 1 as expected
3) Switch to AC input 2
4) ```/Ac/State/IgnoreAcInput``` updates to 0 as expected
5) Switch to AC input 1
6) AC input 1 is ignored however ```/Ac/State/IgnoreAcInput``` stays at 0

**Reason:**
The Multi reports the "ignore AC input" status of the **active** input (through a RAMVAR). Since AC input 1 is ignored the active input remains AC input 2

**Todo:**
- [ ] Fix ignore AC input status in Multi firmware
- [ ] Add new status bits to mk2-dbus

vejpasop, 2020-07-14:
There is an other issue with /Ac/State/IgnoreAcInput, It returns the state active input from the L1 master only. I thought it was a global state. In practice it should be read from all three phases. 

Since currently nobody uses this path yet I propose to remove it for now. (Not include it in v2.60)

Note:
Multi firmware >475 supplies the ignore AC input status per input with the extended status bits. 
With the coming release of mk2-dbus (V3.10, adds grid alarm fix and high DC alarms)  we get the following path.

/Devices/0/ExtendStatus/IgnoreAcIn1 
/Devices/0/ExtendStatus/IgnoreAcIn2
/Devices/1/ExtendStatus/IgnoreAcIn1 
/Devices/1/ExtendStatus/IgnoreAcIn2
/Devices/2/ExtendStatus/IgnoreAcIn1 
/Devices/2/ExtendStatus/IgnoreAcIn2

vejpasop, 2020-11-12:
Added ```Ac/State/IgnoreAcIn1``` and ```Ac/State/IgnoreAcIn2``` 
Path are available with Multi firmware version >= 475
mk2-dbus v3.16

mpvader, 2020-11-17:
Included in v2.62~2

closed 2020-11-17 by mpvader