victronenergy/venus #667: Port python code to python3

izak, 2020-07-02:
Python projects that need porting to python3

- [x] velib-python
- [x] systemcalc
- [ ] dbus-characterdisplay
- [x] dbus-digitalinputs
- [x] dbus-modbus-client
- [x] dbus-mqtt
- [ ] dbus-recorder
- [x] dbus-tools
- [x] vrmlogger
- [x] localsettings
- [x] mqtt-rpc
- [x] dbus-generator
- [ ] dbus-modem
- [ ] dbus-imt-si-rs485tc
- [ ] venus-button-handler
- [ ] dbus-paygo
- [ ] dbus_vebus_to_pvinverter
- [ ] dbus-fzsonick-48tl
- [ ] dbus_pump
- [ ] vesmart-server

Add others to the list

izak, 2021-06-21:
Note for packages required to run this on venus 2.70.

```
opkg install python3-modules python3-pygobject python3-dbus python3-requests
```

jhofstee, 2021-06-25:
I will add my own checkboxes here as well...
Python projects that are included in a venus develop version...

- [x] velib-python
- [] systemcalc
- [] dbus-characterdisplay
- [] dbus-digitalinputs
- [] dbus-modbus-client
- [] dbus-mqtt
- [] dbus-recorder (handle unicode on dbus correctly)
- [] dbus-tools
- [] vrmlogger
- [] localsettings
- [] mqtt-rpc
- [ ] dbus-generator
- [ ] dbus-modem
- [] [dbus-imt-si-rs485tc](https://github.com/victronenergy/dbus-imt-si-rs485tc)   (  <--- modbus, so perhaps best integrated into dbus-modbus-client instead)
- [] venus-button-handler
- [ ] dbus-paygo
- [ ] dbus_vebus_to_pvinverter
- [ ] dbus-fzsonick-48tl
- [ ] dbus_pump
- [ ] vesmart-server (Done, but needs [UTF8 improvements][1] to VictronConnect)

mpvader, 2021-08-17:
CLosing. Part of v2.80; all in; apart  from some bugfixing for which we don't need this issue.

closed 2021-08-17 by mpvader