victronenergy/venus #676: gui memory issue

jhofstee, 2020-07-14:
As reported here https://community.victronenergy.com/comments/55615/view.html and for earlier versions as well, the gui can run out of memory which make running a process, like reboot or check update fail:

```"Resource error (fork failure): Cannot allocate memory"```

Not being able to return from the overview to the pages seem to be caused by this:

```
2020-07-12 12:20:49.219642500 file:///opt/victronenergy/gui/qt-components/qml/Qt/labs/components/native/PageStack.qml:295:7: QML ScriptAction: file:///opt/victronenergy/gui/qt-components/qml/Qt/labs/components/native/PageStack.qml:137: TypeError: Result of expression 'page' [null] is not an object.
2020-07-12 12:20:49.219671500     		if (page.status !== undefined) {
```

I wouldn't be surprised if that is caused by the same reason, failing to allocate the page and hence pushing a undefined page on to the stack.

Why this setup runs out of memory is unknown at the moment, but it must have to do with user interaction or something otherwise we would have reached lots of complaints about the gui getting stuck / ignoring commands.

jhofstee, 2020-07-14:
There was already a issue for this, just not marked for 2.60, https://github.com/victronenergy/venus/issues/653

jhofstee, 2020-07-21:
a ccgx running v2.57 and with a gui not using a lot of memory also encountered this:
```
2020-07-20 17:38:20.731338500 file:///opt/victronenergy/gui/qt-components/qml/Qt/labs/components/native/PageStack.qml:295:7: QML ScriptAction: file:///opt/victronenergy/gui/qt-components/qml/Qt/labs/components/native/PageStack.qml:137: TypeError: Result of expression 'page' [null] is not an object.
2020-07-20 17:38:20.731369500     		if (page.status !== undefined) {
2020-07-20 17:38:56.476809500 file:///opt/victronenergy/gui/qml/OverviewHub.qml:37: Unable to assign [undefined] to QString iconId
2020-07-20 17:38:56.478249500 file:///opt/victronenergy/gui/qml/OverviewHub.qml:23: Unable to assign [undefined] to QString title
```

The device can no longer leave the overview and show the pages.

jhofstee, 2020-07-27:
The latest Venus version has some counters / debug options and is already at the customer its site. There are hundreds of svg's which have a width of 145px and various different heights. Seems like a svg with a property animation on height. In total 30MB or so, so that at least explains where the memory is going (and that is already fixed, since the cache for inline svg's is completely disabled now).

It will now also exit on OOM, so next time it will hopefully occur during allocation of all these images, instead of a random different place (the check update in case).

It remains interesting why the fork fails on low mem, since it doesn't seem to be on memory allocation, not on mmap...

page.status !== undefined is likely unrelated¸ since it occurs on v2.57 as well.

mpvader, 2020-07-29:
Solved - closing

closed 2020-07-29 by mpvader

jhofstee, 2020-07-30:
The fork failure is caused by https://github.com/victronenergy/linux/blob/ccgx_b4.19.74/mm/util.c#L662

Note: that whole check got replaced in https://github.com/victronenergy/linux/commit/8c7829b04c523cdc732cb77f59f03320e09f3386