victronenergy/venus #681: Temperature sense sometimes not forwarded to Multi

izak, 2020-07-22:
From initial investigtion, we're bitten again be velib_python's implementation of `seen()`. Sometimes it returns False for paths that do actually exist. This causes forwarding to be skipped. The problem also goes away when you restart systemcalc, so it appears to be some kind of race condition between service startup when the GX boots.

In this line of code, `seen()` returns False incorrectly:
```
       if vebus is not None and vebus != sense_temp_service and self._dbusmonitor.seen(vebus, '/BatterySense/Temperature'):
           self._dbusmonitor.set_value_async(vebus, '/BatterySense/Temperature',
               sense_temp)
           written = 1
```

izak, 2020-07-22:
Confirmed. If mk2-dbus starts after systemcalc (as it does), it does not correctly record "seeing" this path. Reproducing is as simple as restarting mk2-dbus after systemcalc.

izak, 2020-07-22:
When mk2-dbus starts it does not immediately add all its paths. Some paths are only added to dbus later. Eg:

TemperatureSense available:
```
root@beaglebone:~# dbus -y com.victronenergy.vebus.ttyO5 / GetValue | grep BatterySense/Temperature
 'BatterySense/Temperature': [],
```

But very soon after startup it is not:
```
root@beaglebone:~# svc -t /service/mk2-dbus.ttyO5; sleep 6; dbus -y com.victronenergy.vebus.ttyO5 / GetValue | grep BatterySense/Temperature
```

velib_python scans all available paths as soon as it sees the service, and never again unless a PropertiesChanged is received, so it is unaware that this path exists.

But mk2-dbus does not send a PropertiesChanged when /BatterySense/Temperature becomes available (probably by design). So that breaks the seen() call in velib_python.

Same issue "fixed" (another way) in systemcalc 2.46, when discharge current/voltage was not copied to the Multi.

This is the only remaining place where `seen` is used on a path to the Multi.

Options:
1. Fix the `seen` call. It seems the seen state of a path is not updated after a GetValue call, so it might be easy.
2. Don't use the seen call and simply always write it, but that increases dbus traffic and the control flag indicating that tsense is active will then be incorrect (not shown on the GUI though, so may not matter).

jhofstee, 2020-07-26:
@izak obviously option 1 is the right choice, since mk2_dbus voids the api. Also dbus-spy etc won't display such paths without mk2_dbus behaving correctly. mk2_dbus v3.12 should send the propertiesChanged. At least that now works with dbus-spy. I reverted the systemcalc patch. Can you check DVCC in v2.60~35 to make sure it works correctly now.

closed 2020-07-27 by jhofstee