victronenergy/venus #683: Update vedirect-interface to use new libevent

dpackham, 2020-07-22:
Trying to install venus code on a new raspberry pi install on a pi3b+

I get the following errors.  I cant run jesse on the pi3b+ as it will not boot due to missing firmware

root@raspberrypi:~# apt install vedirect-interface
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 vedirect-interface : Depends: glibc-deb but it is not installable
                      Depends: libevent-2.0-5 (>= 2.0.21-stable-2+deb8u1) but it is not installable
                      Depends: libevent-pthreads-2.0-5 (>= 2.0.21-stable-2+deb8u1) but it is not installable
E: Unable to correct problems, you have held broken packages.

dpackham, 2020-07-22:
vrmlogger installed just fine and runs.  just the vedirect-interface did not

jhofstee, 2020-08-02:
Hi, can you make a issue for this here, https://github.com/jhofstee/meta-bin-deb? The Debian packages is not a Victron project.

closed 2020-08-02 by jhofstee