victronenergy/venus #691: Systemcalc: An Phoenix Smart charger output misses in battery lisr

mpvader, 2020-07-29:
Which affects the html5 app.

https://community.victronenergy.com/answers/57266/view.html

Izak already (most likely) found it.

izak, 2020-07-29:
Systemcalc (via velib_python) tracks the values of dbus paths using the PropertiesChanged events. For some reason `/Dc/1/Voltage` comes up with a voltage followed immediately by an invalidation. When tracking changes using dbus-monitor this is observed:

```
signal time=1596041194.275658 sender=:1.445 -> destination=(null destination) serial=18 path=/Dc/1/Voltage; interface=com.victronenergy.BusItem; member=PropertiesChanged
   array [
      dict entry(
         string "Value"
         variant             double 13.2
      )
      dict entry(
         string "Text"
         variant             string "13.20V"
      )
   ]
signal time=1596041194.279672 sender=:1.445 -> destination=(null destination) serial=20 path=/Dc/1/Voltage; interface=com.victronenergy.BusItem; member=PropertiesChanged
   array [
      dict entry(
         string "Value"
         variant             array [
            ]
      )
...
```

Because systemcalc now things the path is invalid, the battery measurement is not published to dbus/mqtt and the html5 app can't show it.

mpvader, 2020-07-30:
As discussed with Martin, this is no issue for v2.60 since it only happens on beta firmware of the Phoenix Smart Charger.

I’ll move this issue to v2.70.

izak, 2020-07-30:
Description of the issue.

In vetext.c a function called `bmvAuxInputHandling` which deals with the multiple functions the aux on a BMV has and ensures the non-used ones are invalidated. It's executed for everything.

The Smart charger now reports temperature (in the latest beta firmware), so this causes VM, VS and DM to be invalidated. VS (starter battery) is the one affecting us.

VS maps to the same vreg as V2, so this has the effect that /Dc/1/Voltage is incorrectly invalidated.

Suggested fix: Only call `bmvAuxInputHandling` for BMVs.

closed 2020-09-11 by jhofstee