victronenergy/venus #695: ssh-tunnel hanging

jhofstee, 2020-08-05:
As seen on a ccgx of a client, ssh-tunnel can hang for quite a while:

```
2020-08-03 12:23:45.862890500 SSH exited from a successful tunnel, so we exited 'normally' and will not try other ports
2020-08-03 12:23:45.864792500 Terminate ssh tunnel with PID 1609
2020-08-03 12:23:48.663132500 Trying port 80. SSH tunnel process has PID 21689
2020-08-03 12:23:48.666115500 Trying to obtain SSH tunnel port from /tmp/tmp.uwh3wU
2020-08-03 12:24:21.159794500 Trying port 443. SSH tunnel process has PID 21889
2020-08-03 12:24:21.166810500 Trying to obtain SSH tunnel port from /tmp/tmp.uwh3wU
2020-08-04 15:45:41.115468500 Trying port 22. SSH tunnel process has PID 3339
2020-08-04 15:45:41.117606500 Trying to obtain SSH tunnel port from /tmp/tmp.uwh3wU
2020-08-04 15:46:13.703474500 No SSH port allocated. Termnating.
2020-08-04 15:46:13.704232500 Terminate ssh tunnel with PID 3339
2020-08-04 15:46:16.303529500 Trying port 80. SSH tunnel process has PID 3548
2020-08-04 15:46:16.310291500 Trying to obtain SSH tunnel port from /tmp/tmp.F3SQSu
2020-08-04 15:46:48.886748500 Trying port 443. SSH tunnel process has PID 3749
2020-08-04 15:46:48.904095500 Trying to obtain SSH tunnel port from /tmp/tmp.F3SQSu
2020-08-04 15:47:09.664711500 SSH tunnel runs on remote port 59404 and PID 3749 is connected to 84.22.107.120; storing in dbus
```

wiebeytec, 2020-08-11:
The assessment of the fix, that it waits forever on `wait $ssh_pid` is correct with the log output: that it goes from port 443 to 22 without other text lines, and still using `/tmp/tmp.F3SQSu`. I haven't done testing, but it seems to be a good patch, with the minor caveat that it's theoretically possible that `$ssh_pid` is reclaimed by another process during a race condition (I know, a wrap-around in PIDs is unlikely).

jhofstee, 2020-08-20:
closing this, it was left open to discuss it with Wiebe.

closed 2020-08-20 by jhofstee