victronenergy/venus #697: MFD App on/off issue

mpvader, 2020-08-16:
https://community.victronenergy.com/answers/57649/view.html

Needs triaging - Reproducible? Regression?

martinbosma, 2020-08-17:
I'm unable to reproduce with v2.58, v2.60~38 and with the MFD App version from v2.60~37 (since I can't install v2.60~37).
Changing the mode from the MFD App reflects in the GUI and visa versa.

mpvader, 2020-08-17:
ok I've asked the customer for more info.

martinbosma, 2020-08-20:
[Update from customer](https://community.victronenergy.com/questions/57252/venus-os-v26037-available-for-testing.html?childToView=59352#comment-59352); same issue with v2.60~38. He posted a screenshot (below) of the boat & motorhome overview, which I didn't test before.

[image]

However, even with the overview it is working perfectly when I tested it (on a real multi). So the GUI, overview and MFD app are updated accordingly when changing the multi to on/off/charger only. 

In above screenshot the AC loads are 0W, while in my case this goes to `---` shortly after the multi switches to Off. Perhaps because he has a different setup than I have?

vejpasop, 2020-08-20:
@martinbosma I can only get the GUI in this state through the Remote console by clicking on the AC-Mode and cycle to ON.
If then I switch to the MFD, in an other tab of my browser, and switch the Multi OFF. The AC-Mode control remaines ON,

The pencil icon in the AC mode field is visible in this case. (It is also visible in the image from the customer)
[image]

martinbosma, 2020-08-20:
@vejpasop Yes, you are right! It looks like the AC Mode is not updated when in edit mode (pencil icon). 

I verified with dbus-spy that `/Mode` is updated when I change the mode on the MFD app, but the overview is not updating accordingly. It seems to me that this is an issue related to the overview, not the MFD app.

closed 2020-08-24 by jhofstee