victronenergy/venus #707: Consumption not shown on VRM with systems that has no Inverter/Charger

izak, 2020-09-21:
This does not always happen. It seems to affect:

1. Raspberry Pi
2. No VE.Bus inverter/charger, so the system only has DC consumption values
3. BMV used to measure the battery.

The bug happens because of a race condition. Once systemcalc finds the BMV and makes it available as the selected battery monitor, the energy counters may not be available on dbus yet. If vrmlogger reacts to the available battery monitor within this small window, the energy counters will be ignored and no consumption data will be available.

izak, 2020-10-12:
Problem is back again. Moving set_seen earlier helped but there is still a race condition somewhere and sometimes the history paths (used to identify a BMV 700 or later) are still not there by the time vrmlogger looks for them.

izak, 2020-10-12:
OK, this is what happens.

1. systemcalc and vrmlogger shows up on dbus (order probably doesn't matter).
2. Battery monitor shows up, systemcalc scans it.
3. vrmlogger sees /ActiveBatteryService change
4. vrmlogger tells its kwhdeltas component to use the new battery monitor
5. Since vrmlogger has not reacted to the NameOwnerChanged yet (and scanned the service), it cannot find the energy paths on dbus. It rejects it.
6. Finally, after rejecting the battery monitor, vrmlogger scans the service.

Depending on the order of steps 5 and 6, things work or they don't.

Possible solutions:

1. Redo battery selection when a .battery service shows up (probably easiest)
2. Instead of relying on the dbusmonitor object (which has not yet scanned the available paths), query it directly (probably more correct and cleaner).

izak, 2020-10-15:
OK, this is what happens.

1. systemcalc and vrmlogger shows up on dbus (order probably doesn't matter).
2. Battery monitor shows up, systemcalc scans it, sets /ActiveBatteryService accordingly.
3. vrmlogger sees /ActiveBatteryService change
4. vrmlogger tells its kwhdeltas component to use the new battery monitor
5. Since vrmlogger has not reacted to the NameOwnerChanged yet (and scanned the service), it cannot find the energy paths on dbus. It rejects it.
6. Finally, after rejecting the battery monitor, vrmlogger scans the service.

Depending on the order of steps 5 and 6, things work or they don't.

But it gets worse.

1. The energy paths `/History/DischargedEnergy` and `/History/ChargedEnergy` are not immediately fetched by vedirect-interface and show invalid until the information is received, they may also only show up later. Usually they show up very quickly for a BMV. But...
2. A BMV isn't the only kind of battery monitor. Some can-bms batteries also support it now, and the time it takes to receive the CAN-frame with this info leaves an even larger window for a race condition.

Possible solutions:

1. Redo battery selection when a .battery service shows up (probably easiest, but solves only half the problem)
2. Just redo battery determination every time the deltas are calculated.
3. Maybe something less invasive than a complete rewrite of the mechanism?

mpvader, 2020-11-15:
Looks like there is a bug in the code. Seen on v2.62~1 on CCGX in CI:

```
2020-11-15 08:20:50.688425500 MainThread-dbusmonitor: Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
2020-11-15 08:20:50.698832500 MainThread-dbusmonitor:        com.victronenergy.battery.ttyUSB0 has device instance 288
2020-11-15 08:20:50.996268500 MainThread-dbusmonitor: ===== Search on dbus for services that we will monitor finished =====
2020-11-15 08:20:51.208892500 MainThread-kwhdeltas: active battery service: com.victronenergy.battery.ttyUSB0
2020-11-15 08:20:51.215761500 Traceback (most recent call last):
2020-11-15 08:20:51.216090500   File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 706, in <module>
2020-11-15 08:20:51.217785500     main()
2020-11-15 08:20:51.218199500   File "/opt/victronenergy/vrmlogger/vrmlogger.py", line 628, in main
2020-11-15 08:20:51.219594500     kwhdeltas.start(dbusMonitor)
2020-11-15 08:20:51.219971500   File "/opt/victronenergy/vrmlogger/kwhdeltas.py", line 203, in start
2020-11-15 08:20:51.220553500     self.handle_new_service(name, instance)
2020-11-15 08:20:51.220881500   File "/opt/victronenergy/vrmlogger/kwhdeltas.py", line 271, in handle_new_service
2020-11-15 08:20:51.221519500     if self._battery_service  is not None and self._battery_service.service == service:
2020-11-15 08:20:51.221939500 NameError: global name 'service' is not defined
```

closed 2020-11-19 by jhofstee