victronenergy/venus #709: BatteryLife/ESS: Scheduled Charging priority changes reduces battery life

Yoda101, 2020-09-23:
I believe the change to the new Venus V2.60 software changed the priority of adhering to scheduled charging with the consequence that the batterylife is reduced. In South Africa we have lots of rolling black-outs called loadshedding. 
The previous priority placed scheduled charging above self-consumption which resulted in a single charge/discharge cycle per day (see figure-1). The new priority placed scheduled charging below self-consumption which resulted in a 3 or 4 charge/discharge cycles per day (see figure-2). **This will severely reduce the battery life**.

My daily scheduled charging using a combination of Solar and Grid power is as follows (see figure-3):
1. Starts at 9h00 to 15h00 (SOC up to 30%) to allow solar charging,
2. Then 15h00 to 24h00 (SOC to 95%) to have a full battery available, 
3. Then 0h00 to 5h00(SOC  95%), 
4. then off for daily discharging battery from 5h00 to 9h00 
5. when the cycle repeats.

The original priorit resulted in battery backup available daily from 9h00 to 5h00 the next morning with a single discharge cycle and the best battery life .
Please revert this priority or provide an option to select this mechanism
[image]
[image]
[image]

izak, 2020-10-02:
No changes were made to scheduled charging on Venus 2.60, there is nothing we can revert. If you think something is wrong, please provide us with a VRM portal id so we can investigate and contact you.

Yoda101, 2020-10-04:
Hi,
Thank you for your response -
VRM Portal ID:

f045da7d75b0

Best regards/Groete.
Jan Hurter

Mobile: +27 82 938 7582 - Home: +27 82 826 2027
E-Mail: Jan.Hurter@gmail.com

Yoda101, 2020-10-04:
Hi,

Unless my assumption is incorrect that the "Scheduled Charging" should have
a higher priority than the "Discharge Cycle"

Best regards/Groete.
Jan Hurter

Mobile: +27 82 938 7582 - Home: +27 82 826 2027
E-Mail: Jan.Hurter@gmail.com

izak, 2020-10-06:
Thank you. Someone will contact you.

closed 2020-10-06 by izak

Yoda101, 2020-10-06:
Why is this issue closed?

izak, 2020-10-07:
I sent you an email to the address listed for the user for the site you referenced. There is nothing wrong with scheduled charging. Your AC input limit is set too low. That's why the issue us closed.

Yoda101, 2020-10-08:
Izak,,

Thank you for all the help!
Best regards/Groete.
Jan Hurter