victronenergy/venus #713: "Create Access Point" option missing on Venus OS on Raspberry Pi

Kradenko, 2020-10-01:
Hi 

"Settings → Wi-Fi → Create access point" isn't available on the Venus OS version 2.60 for the Raspberry Pi

I have seen one other thread that had no replies regarding this: https://community.victronenergy.com/questions/48565/access-point-option-missing-venus-os-raspberry-pi.html

Does anyone know why this option isn't available? Or if there is a config setting on the Pi I need to do. The Raspberry Pi 2 has a wifi dongle that does pick up other networks. Pi 3 I used the built in Wifi to connect to other networks.

Kind Regards,

izak, 2020-10-08:
The platforms that have an access point all have two Wi-Fi interfaces, so that one can be used to create an access point, and the other can be used to connect to a Wi-Fi network. Since the Raspberry Pi has only one wireless interface (in selected models), the AP feature cannot be supported without considerable extra effort (to share one Wireless interface between the functions).

closed 2020-10-08 by izak

Kradenko, 2020-10-08:
Thank you izak, appreciate the reply. If I plugged in an additional wifi dongle into the pi 3 (Which already has build it wifi) could it work? Or is it just outright disabled on the VenusOS build?

Kind Regards,

izak, 2020-10-08:
The software needed to make it work is called hostapd, and we simply don't include it in the builds for platforms where the hardware isn't known to exist (including our own CCGX platform). It's outright "disabled" in the sense that the software is not included at all.

To make it work, you'd have to (cross) compile and install hostapd, and then configure it to use the right interface. Of course we could do it, but there really isn't much demand for that, nor do we have the time.

OptiMysticall, 2021-11-13:
If you're not needing anything besides simply making the raspberry pi an Access Point then simply open the shell or plug a monitor to it and do this:  connmanctl tether wifi on VENUSAP 12345678 

Note the command will say it requires an AP Passphrase cause it's not long enough. 

To persist this at reboot edit this file with nano: 
nano /etc/connman/main.conf

and add this to the bottom:

TetheringTechnologies = wifi
PersistentTetheringMode = true
AddressConflictDetection = true

Credit goes to pagedo in this posting: https://community.victronenergy.com/questions/48565/access-point-option-missing-venus-os-raspberry-pi.html

HolgerCR, 2023-04-02:
I activated the AP on my Raspberry i can see the AP now and also connect to them my question is if its not possoble to use the venusap for internet connection ? my device shows no internet connetion when i connect the venusap

mpvader, 2023-04-02:
Hi, no thats not possible.

in Venus OS we didn’t include a router function.

for any further discussion on this, could you please  use Modifications section on Victron Community Forum?

these messages here trigger emails to many Victron people, and adding or supporting a router function is not something we’re interested in. Sorry.