victronenergy/venus #718: No AC Input

Barpo, 2020-10-27:
Hi,

Im rather new in venus pi. I connected a CG powermeter EM111. I can see the grid meter readings like AC Phase L1.
How come I do not get AC Input?

mvg Barry

[image]

Yuphorya, 2020-12-05:
Interested if you got this working? I plan to do the same.

Wholistic, 2020-12-07:
Hi @Barpo and @Yuphorya,

The AC input display requires a Victron Inverter Charger to be connected, there is some more info in this Victron Community discussion here - https://community.victronenergy.com/questions/46152/possible-for-cerbo-gx-to-show-incoming-power-witho.html

This is working as intended. 

Any further conversation on this would probably be best located in the modifications space here - https://community.victronenergy.com/spaces/31/index.html

closed 2020-12-07 by mpvader