victronenergy/venus #719: vecan-dbus floods dbus

izak, 2020-10-28:
Observed at a site with 24 VE-Can solar chargers, running Venus 2.60. Around noon the vecan-dbus process starts to flood dbus with a hundred messages per second. Diagnosed with my own dbus-top tool:

[image]

[image]

These paths appear to cause all the traffic:

[image]

Affected [site 1][1] and [site 2][2].

[1]: https://vrm.victronenergy.com/installation/73577/advanced
[2]: https://vrm.victronenergy.com/installation/73940/advanced

jhofstee, 2020-11-16:
On a system with 24 MPPTs, there can be at least 88 msg / sec. 24 for the Yield/Power and roughly 16 / sec for the Dc and Pv voltage and current. So the dbus-top output shows what can be expected.

jhofstee, 2020-11-16:
on site 2:

```
root@einstein:~# ip -details -statistics link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 100
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 100 
	  bitrate 250000 sample-point 0.875 
	  tq 250 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  sun4i_can: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 24000000
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  15         8063867    4270770    89574      468481     15        numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    723382392  627357376 8063879 161     12      0       
    TX: bytes  packets  errors  dropped carrier collsns 
    177705391  22214367 4270770 15      0       0 
```

jhofstee, 2020-11-16:
site 1:

```
root@einstein:~# ip -details -statistics link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 100
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 100 
	  bitrate 250000 sample-point 0.875 
	  tq 250 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  sun4i_can: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 24000000
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          23743      152058     0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX: bytes  packets  errors  dropped overrun mcast   
    331311743  41421111 23746   81      3       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    11635915   1454535  152058  0       0       0 
```

closed 2020-12-01 by jhofstee