victronenergy/venus #721: Feature request - relay control based on selectable temperature sensor

Wholistic, 2020-10-29:
Much like how you can start a pump using the relay based on a tank level sensor reading, 

Would be nice if you could open or close the relay based on a temperature reading. 

For example, if temp gets >30 degrees, start the air conditioner or fans, until temp drops below 25 degrees. 

Of course - this can already be done via Node Red (sort of).

nmbath, 2020-12-21:
Not sure if this is of any help, but here is a dbus service to control a relay based on temperature probes. Very similar to the Pump dbus service. I also copied the test harness from the pump service and it passes!

I also modified the velib_python file vedbus.py to use /Mgmt rather than /management

I have enumerated the relays so it can be applied to either relay 1 or 2.

If this can be used to speed up delivery a number of people in the community would be very pleased.

[dbus-temperatureSwitch.tar.zip](https://github.com/victronenergy/venus/files/5725601/dbus-temperatureSwitch.tar.zip)

nmbath, 2021-01-03:
Further to my previous comment attached is a opkg package that will modify the system allowing temperature control of the relays. It also includes some changes to the dbus-digitalinputs to support input types from a BMS for Allow to Charge, Allow to Discharge and Pre Alarm.

The relay can be associated with any temperature source, either an analogue temperature input or a battery monitor providing /Dc/0/Temperature.

If this is of use please do use.

[MB-Modifications_0.1_cortexa7hf-neon-vfpv4.ipk.zip](https://github.com/victronenergy/venus/files/5762010/MB-Modifications_0.1_cortexa7hf-neon-vfpv4.ipk.zip)

jepefe, 2021-12-10:
This feature has been implemented and added to the ToDo for including it in Venus.

The implementation supports two conditions per temperature sensor where each one consist on a temperature range (high or low) where the configured relay must be closed.

For example a temperature sensor in the battery cabinet can be used to switch on a fan in case of high temperature or a resistor to keep a Lithium battery above the cut-off temperature:

Condition 1
Activation temperature: 35ºC
Deactivation temperature: 25ºC
Relay: 1

Condition 2
Activation temperature: 4ºC
Deactivation temperature: 10ºC
Relay: 2

Both conditions can be set to activate the same relay. This might be useful to switch on an air conditioner (set to auto) in case of high or low temperatures.

Multiple sensors can control the same relay. Relay will keep closed as long as there are one active condition of any sensor.

Conditions can be configured in two places. 
  1. From the temperature sensor menu: Devices list -> Temp sensor -> Setup -> Relay

[image]
[image]

  2. From the relay settings page(Settings -> Relay -> Temperature sensors), here a the sensors are listed showing the condition values to check them all in a glance. This menu is only shown if at least one of the relays is configured as a temperature relay.
  
[image]
[image]
 [image]
[image]

mpvader, 2021-12-10:
Great!

closed 2021-12-17 by mpvader

lennycb, 2021-12-30:
@jepefe What has to be done to use this with the internal CPU temp sensor from the Pi4 or a DS18B20 1-wire sensor.
According to the description, entries for the sensors and their values should be in the "Temperature sensors" tile. But is missing with me. What did i forget?
[image]
[image]
[image]
[image]

kwindrem, 2021-12-30:
This should be better discussed in the community forum.

I recall seeing a mod to make the RPI temp sensor visible. I tried it and worked fine but has not been upgraded for V2.80 (Python 3).

lennycb, 2021-12-30:
Hi @kwindrem, I have no problem with the ad. I've already adjusted the mod. See forum:
[https://community.victronenergy.com/questions/58792/raspberry-pi-3b-heat-temperature.html?childToView=113688#comment-113688](url)

So, they are displayed as they should, but unfortunately I cannot control relays with the sensors. I'm not sure if this is only intended for the RuuviTags.

kwindrem, 2021-12-31:
Made a couple of tweaks to LHardwick's code to:

1) make it run on both python 2 and 3

2) make the CPU temp alarm relays available 

The second problem was that the dbus service name included a dash (-) which caused com.victronenergy.temprelay to crash. Changed the - to _ and the alarm settings show up in the CPU temp sensor Setup menu.
[dbus-i2c.py.zip](https://github.com/victronenergy/venus/files/7797001/dbus-i2c.py.zip)

lennycb, 2022-01-01:
Great, works. I hadn't thought of a change for Python2 and Python3. Will become obsolete after an update to Venus> v2.80.

What error are you seeing here?
# --------------------- was causing an error
#from w1thermsensor import W1ThermSensor
# ---------------------
I had to activate it again to read my DS18B20.

This library is normally not included and must be installed later.
To do this, I first had to install PIP, then the library.

kwindrem, 2022-01-01:
The error was that the import file could not be found but that makes sense based on your explanation.

I wrapped a try: except: pass around the import statement so it no longer fails:

[dbus-i2c.py.zip](https://github.com/victronenergy/venus/files/7797972/dbus-i2c.py.zip)

Here's my setup script that uses SetupHelper if you are interested:

[RpiTemp.zip](https://github.com/victronenergy/venus/files/7797978/RpiTemp.zip)

You should be able to add the missing bits for the w1... stuff

To make this work with auto download and install, it needs to be available on GitHub. I generally set up a 'latest' tag for the most recent released version but that's optional. The branch/tag in PackageManager can simply be set to 'main'. You may wish to change the package name back to what you are using now.