victronenergy/venus #723: vrm backlog: Prepare for > 4GB files

jepefe, 2020-11-05:
For FAT32 there is a well known maximum file size limit of 4GB. It is feasible that on a busy installation with a lot of devices connected to Venus this limit reached within a few years.

There is already a user in community with a sqlite3 file a bit larger than 500MB that corresponds to only ~6 months of logging. So in 4 years the 4GB limit will be reached, or even earlier if the amount of logged data is increased by connecting more devices or because we add more events to the logging.

To prevent problems we can; a) Split the sqlite3 filre or b) Support exFAT.

After exposing the problem to Matthijs, he decided that the best option is to include exFAT support on Venus.

A year or so, Microsoft freed the patents to make possible to include exFAT support on Linux, so perhaps it is already on main Linux branch or available as a patch.

jepefe, 2020-11-05:
Seems that I have no permissions to assign users to the issue, so I tag you @mansr

mansr, 2020-11-06:
Mainline Linux has exFAT support since v5.7 with some fixes in subsequent releases. I haven't tested it, so I don't know if there are any problems to be expected.

jhofstee, 2021-06-21:
for completeness, it is not supported on rpi's

closed 2021-06-21 by jhofstee