victronenergy/venus #730: Complete new Phoenix Smart Charger support

mpvader, 2020-11-18:
There is an updated version of the Phoenix Smart Charger going into production. This issue is about making sure its fully supported.

- [ ] (mbo) explain whats different compared to previous model
- [ ] (mbo) arrange that one is installed on one of the fixed test setups in Groningen
- [ ] (ibu) check vrm & modbustcp support

martinbosma, 2020-11-18:
The new hardware of the Phoenix Smart IP43 Charger has the following changes compared to the old one:
- Universal mains vs 230V only
- Added AC input current measurement; available on `/Ac/In/L1/I`, see [here](https://github.com/victronenergy/gui/blob/master/qml/PageAcCharger.qml#L84)
- Added AC input current limiting; available on `/Ac/In/CurrentLimit`, see [here](https://github.com/victronenergy/gui/blob/master/qml/PageAcCharger.qml#L40)

martinbosma, 2020-11-18:
@izak I'll ping you once I've installed the charger on one of the test setups; still waiting for final hardware myself. I will install a new and old model.

martinbosma, 2020-11-26:
@izak There was not enough space on Groningen Hub-1 or ESS to mount the chargers, so I've created a [new](https://vrm.victronenergy.com/installation/86854/dashboard) system.\
It has both an old and new charger connected to it.

izak, 2020-11-27:
modbus-tcp and vrm support already sufficiently in place.

mpvader, 2021-06-28:
@martinbosma  can you confirm that all works OK for both the old & the new version of the phoenix smart charger?

I know the overview page doesn't work well; thats not part of this issue; this is just about the entry into the device list etc.

martinbosma, 2021-06-28:
>can you confirm that all works OK for both the old & the new version of the phoenix smart charger?

I can confirm it works as expected on Venus v2.71

closed 2021-06-28 by mpvader