victronenergy/venus #735: Feature request: Backup/transfer configuration using USB/SD-card

izak, 2020-11-23:
Add a feature whereby the configuration of a GX device can be backed-up to offline media, such as an SD-card or USB flash drive. Also add a restore feature, so that configuration can also easily be cloned.

Caveat: Some configuration is stored under the serial number of a device, and copying it won't apply on another GX device.

- [ ] Implement backup feature
- [ ] Implement restore feature
- [ ] Determine what works and what doesn't (re serial number specific items) and document it
- [ ] Document feature

izak, 2021-02-02:
## Files to backup
```
/data/conf/settings.xml
/data/conf/vncpassword.txt <--- so new device will have same remote console password
```

Using the menu as a guide. This is what works and what doesn't work.

## What works

* Everything under Settings -> General
* Firmware update preferences under Settings -> Firmware
* Date & Time settings
* Remote console settings
* Everything under System Setup, **except Marine MFD app integration**
* Everything under DVCC
* Display and language settings
* VRM portal settings
* ESS settings
* Modbus TCP devices (if IP addresses of devices don't change)
* Settings related to modbus-tcp devices (but not the device configuration itself)
* GSM modem settings
* GPS settings (format, unit)
* Generator start/stop settings
* Tank pump settings
* Relay settings
* Service configuration, except NMEA2000-out instance numbers which may be different
* Analog and digital input configuration

## What works partially
* PV inverters: IP addresses and whether to scan for PV-inverters

## What doesn't work
* Energy meter settings (role and phase configuration, DeviceInstance number)
* Ethernet EM24 meter and Smappee meter DeviceInstance, role and CustomName).
* PV-inverter settings (position and phase configuration, DeviceInstance number)
* Ethernet configuration is not preserved
* Wi-Fi configuration is not preserved
* Allocated DeviceInstance numbers for VE.Can solar chargers
* CustomNames for most devices (vague, but this is almost always bound to the serial number of the device if not stored on the device itself).

jhofstee, 2021-06-17:
> /data/conf/vncpassword.txt <--- so new device will have same remote console password

You can forget about that, the password is salted. If won't work on an other installation by just copying the file.

izak, 2021-07-30:
Backup screen if no SD-card or USB is available:

[image]

Options when there is an SD-card. The export option is meant for transferring to another device (it removes device specific content from the file so you don't end up with grid meters and things you don't have on the other side).

[image]

Operation is communicated using toasts as always:

[image]
[image]

kwindrem, 2021-07-30:
Most excellent addition.

For backup/restore to the same or replacement GX device, the serial numbers of connected devices will not change. If serial numbers don't change it should be possible to retain custom names. Yes?

For export, will this write to SD/USB memory for transfer to another device or will this happen over the network?

When SD/USB is inserted into a GX device, it immediately assumes the role of offline VRM storage. This is probably not a good idea if the memory device is being used for settings transfer. If on-line, no data is stored to the memory device but if off-line VRM data will be sent to the card and will therefore be lost.

kwindrem, 2021-09-14:
Regarding the interaction with VRM off-line storage:

Would it be appropriate to ask first? That is, pop up a dialog when a memory device is inserted asking if the card should be used for off-line VRM storage.

This would require an extra step when inserting a memory device, but how often does this actually happen?

Your idea of moving VRM data back to to internal storage would work only if there is sufficient space on internal storage and if there are a lot of records could take some time. But again, dialogs could allow the user to do what suits their situation best: Most recent records, no records; then ... transferring data please wait ...

mpvader, 2021-12-15:
After being shortly added to v2.80, this was removed since it needs more work. Moved it to the v2.90 milestone just now.

kwindrem, 2021-12-15:
Sorry, I missed the question back in Sept:

Yes, asking first to begin logging to external media would solve most of the conflicts. You are right that this would not normally happen often: only when inserting the media for the first time and after reinserting it presumably to transfer the contents to VRM via your computer.

If we can get the system to only log to external storage when given permission, I'm not sure how important copying back to internal on eject would be. This is really only an issue when the logger automatically uses external storage.

I also posted today in the Mods forum that the eject removes ALL external media. That probably needs to be handled differently as well.

Along with asking first to use external media, it would be nice if you had a choice of drive if more than one is inserted. We can name drives in Windows and Mac OS but not sure how many get renamed from Untitled or whatever Windows does. Not sure how difficult it is to get to these names from Linux.

A way to avoid logging to the "wrong" media would be to add a flag file to the drive, the presence of which would be used to enable logging. You could set this during the process of asking to use the drive for VRM backup, and you could look for the flag's presence when booting. The prompt when inserting media could change to "Continue using this external drive for VRM" or whatever if the flag existed.

mpvader, 2021-12-16:
hi, yes adding flag files is most likely what will be done

lfhoward, 2024-03-12:
I am wondering if any progress has been made on this since 2021. It would be a fantastic feature.

charlesrg, 2024-10-12:
looking forward for this to work. Backup/restore is an essential feature.

Wholistic, 2025-02-18:
There was another post on community requesting an update to this feature here as well - https://community.victronenergy.com/t/move-settings-from-one-cerbo-to-another

KapitainNemo, 2025-05-31:
any progress on these efforts?

gyrex, 2025-06-21:
I'd love to see this progress. Surely this is an essential function for such a pivotal part of Victron's system.