victronenergy/venus #736: DVCC: Also control AC chargers and inverters

izak, 2020-11-23:
* Where a system has AC chargers as well, and where those chargers can be externally controlled, control the charge voltage and current.
  * Carefully consider what happens in system with both AC chargers and Multi: Do we prioritise AC chargers over the Multi?
* In a system with an inverter, also turn off the inverter if the battery sends DCL=0.
  * Carefully consider how this interacts with the user.
    * If the user switches the inverter off while DCL=0, and DCL is subsequently raised, inverter should remain off.
    * If the GX device switches inverter off due to DCL=0, and user manually flips switch on/off, inverter should (probably?) remain off.

TODO
- [ ] Distribute DVCC charge current limit also over AC chargers (eg Phoenix smart charger).
- [ ] Switch off inverter when battery sends DCL=0

izak, 2021-04-13:
Assigning Martin, because there is a question about the Phoenix Smart IP43 Charger. It supports these vregs already:

* VE_REG_LINK_VSET
* VE_REG_LINK_CHARGE_CURRENT_LIMIT
* VE_REG_LINK_NETWORK_MODE
* VE_REG_BAT_MAX_CURRENT

But it does not currently support receiving VE_REG_LINK_VSENSE, VE_REG_LINK_TSENSE, and VE_REG_LINK_BATTERY_CURRENT (SVS, STS and SCS) from both VE.Smart networking and VE.Direct. If this is to be added as well, firmware work will be required.

mpvader, 2021-04-13:
Hey @izak , Martin wont have time for this anytime soon. Pls start with a different charger model: the Skylla IP65

mpvader, 2021-08-17:
Above is included in v2.80~something.

For inverter control, this is now implemented such that systemcalc controls Phoenix VE.Direct Inverter via their normal on/off switch. And for that to properly work it keeps its state: ie don't switch an inverter on if it was first off.

Not good enough. Its needs taking out. This whole issue came up for the Inverter RS, not for these Phoenix VE.Direct Inverters. What I'd prefer is to now take that control for the Phoenix Inverters out, including its keeping state. While keeping the Inverter RS part in.

And then leave it like that: Phoenix Inverters with VE.Direct port are simply not supported at the moment. And then some day we can make it properly, including a fw change of the PHoenix Inverter VE.Direct; but not now since that team has other prios.

izak, 2021-09-22:
Some investigation today: It doesn't look as if the Skylla IP65 supports VE_REG_LINK_NETWORK_MODE, and writing to VE_REG_LINK_VSET is not supported. The charger has firmware `v1.10` installed, which is the latest.

mpvader, 2021-12-15:
@izak what is the status on this? Iirc you worked on it with Martin for the Phoenix IP43 charger rev 2, connected via VE.Direct? (and the rev1 model won't get this due to hardware limitations).

Basically, what I want to know is what has changed between v2.73 and v2.80-now. And what will then remain to be done later. 

Also for changelog and documentation.

izak, 2021-12-16:
I started working on it and found that basically none of our chargers supported the VREGs required for this. Martin then added them for the Phoenix IP43 charger. He also made changes to systemcalc to test it, but I have yet to look at his changes and possibly incorporate them. What needs a bit more thought is prioritising solar over AC chargers, something I don't think has been done.

ThiemoVanEngelen, 2022-01-31:
>Add link parameters to VE.Direct driver for chargers

This is implemented in vedirect-interface v3.67, included in venus 2.80~27.

mpvader, 2022-08-16:
removed from v2.90 milestone; its not finished, and not important enough to delay v2.90 any further either.

chaerle, 2023-02-14:
When will this be integrated? Currently the Multiplus II is not following DCL=0 or CCL=0 but still charging/discharging the battery even though forbidden by the battery. It could drive battery into overvoltage or undervoltage. DCL=0 and DCL=0 must be followed under all circumstances. 

. Also Modbus register 38 is not working like documented. 

https://community.victronenergy.com/questions/186404/multiplus-ii-switch-of-charger-inverter-on-dcl0-or.html

msigman, 2023-06-10:
Also interested in this functionality as I have a Phoenix Smart Charger (VE.Direct HW Rev 2) that I would very much like to get under DVCC control.  I am willing to help code and/or test, with some guidance, if it would speed things along.  Cheers

GregorKopka, 2024-09-27:
One year later... will the €500 IP43 charger get some love?

msigman, 2024-09-27:
> One year later... will the €500 IP43 charger get some love?

+1

hoschid, 2024-10-09:
> One year later... will the €500 IP43 charger get some love?

+1

msigman, 2025-04-06:
> Yes, I would very much like this feature. I found it incredibly difficult to choose an AC charger and ended up with an IP65 which I regret now as it doesnt support DVCC. I get very many errors with it in a DC network like error 65 reported by the MPPTs smartsolar.
> 
> I had to turn off the Watch function and also bulk charge limits for a Lithium NG battery bank (4 x 300ah Lithium NG).
> 
> I am very willing to be a beta tester as well and provide feedback to victron or give access to my systems for your remote viewing.

100% agree.  And also happy to beta test or even write the code if access can be provided

rajlaud, 2025-07-03:
I have a Phoenix IP43 (VE.Direct HW Rev 2) Smart Charger that I would like to get under DVCC control. It's an all-Victron system and quite frustrating that a current model Victron charger doesn't support this. Happy to beta test as well.