victronenergy/venus #740: Cerbo audible alarm not working?

mpvader, 2020-11-26:
https://community.victronenergy.com/questions/68678/cerbo-gx-internal-audible-alarm-not-working.html

mkrenzke, 2020-11-26:
I'd assume it is not a (general) hardware issue since the buzzer should be part of the test procedure. But I will double check it actually is.

mkrenzke, 2020-11-26:
Pedro just confirmed: Testing the buzzer is part of the test plan.

mpvader, 2020-11-26:
It was already double checked to be an issue on at least two Cerbos. (Guy his Cerbo, and this customers Cerbo).

@mansr pls check if you can reproduce or not.

mansr, 2020-11-26:
Configuring a digital input as "door alarm" works as expected on my Cerbo with v2.62~4.

Wholistic, 2020-11-27:
The alarm does make some sound, but it is extremely quiet.

This is an audio recording of the alarm, over the background sound of the Multiplus-II fan.
It is not possible to hear the alarm (to my ear) in the same room as the multiplus fan. 

It looks like the alarm is about 6khz - and you can see when the alarm is sounding (with mic pressed up to the unit), it barely registers over the background fan noise.

https://www.dropbox.com/s/2rftnw4oc3kh08m/cerbo%20gx%20alarm.mov?dl=0

jhofstee, 2020-12-02:
included in the next v2.70~

closed 2020-12-02 by jhofstee