victronenergy/venus #741: Inconsistency in naming structure between dBus Services

nmbath, 2020-11-26:
Good morning,

I am not sure if this is the correct place to raise an issue or if it should have been done on the community forum.

Looking at the QML page for devices, it looks at "item.bind: Utils.path(root.bindPrefix, "/Mgmt/Connection")"

Within the digital interfaces this is actually "/Management/Connection", yet in other areas it is "/mgmt/....", example from my VE.Bus structure "'Mgmt/Connection': 'VE.Bus'"

This means that for at least a digital interfaces, there is no connection displayed. I am not sure if this is intentional or is just a bug. Please could this be addressed.

mpvader, 2020-11-26:
Hi, its supposed to be /Mgmt, as per here:

https://github.com/victronenergy/venus/wiki/dbus#generic-paths

What digital interfaces are you referring to? Thanks for raising this.

nmbath, 2020-11-26:
Good afternoon,

I thought it should be /Mgmt.

This is on Raspberry Pi 2. This is output from dubs-spy:

com.victronenergy.digitalinput.input04                                                    Bilge pump

and then

com.victronenergy.digitalinput.input04
Alarm                                                      0
Connected                                             1
Count                                                     0
CustomName
DeviceInstance                                     4
InputState                                             0
Management/Connection                    /dev/gpio/digital_input_4
Management/ProcessName                /opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py
Management/ProcessVersion             0.7
ProductId                                              41316
ProductName                                       Bilge pump
State                                                      2
Type                                                       3

Am running v2.62~3

I think the issue is with bus_digitalinputs, as it has

        self.service.add_path('/Management/ProcessName', __file__)
        self.service.add_path('/Management/ProcessVersion', VERSION)
        self.service.add_path('/Management/Connection', path)

I modified these three lines to be /Mgmt and now my screens show the driver...

See attached before and after screen shots
[image]
[image]

closed 2020-12-02 by jhofstee