victronenergy/venus #746: Cerbo LEDs disable option

IsaiahJTurner, 2020-11-29:
The LEDs on the Cerbo GX are incredibly bright and penetrate the case itself so even electrical tape covering the lights is ineffective. An option to disable the LEDs should be added. This setting must be reset when the reset button is used.

mpvader, 2020-11-30:
Hi, yes they are indeed very bright. And the flashing is rather "obnoxious", but changing the sequence after market launch also has its disadvantages.

Your suggestion might be a good solution to get it all solved; we'll discuss it.

thank you!

Chrischan-git, 2023-08-20:
is the topic still under discussion? the leds annoy me too. and there are more with each device. feels like some kind of christmas tree in the camper van 😉