victronenergy/venus #747: Addition of Digital Input Types to dbus_digitalinputs.py

nmbath, 2020-11-29:
Due to the open source nature of the code, people area starting to make additions. This is of real benefit to Victron, but does create a problem.

I myself am adding some additional digital input types. I am also aware that Osaether else has added one (https://github.com/osaether/dbus-inverter-ctrl), there maybe others. Managing the merge of various peoples code at a later date or as patches to the main code base for personal use, will become complex and difficult if we all start using the same Type ID. Some means of adding to to the list so that duplicate uses do not occur and cause issues in other parts of the code base would be really useful.

I am adding two, 11 = "BMS Allow to Charge" and 12 = "BMS Allow to Discharge". Osaether has added 10 = "Inverter control"

If these can be added to the list that would be brilliant and save complex code merges at a later date.

mpvader, 2020-12-03:
Izak your comment pls.

izak, 2021-02-02:
Adding additional input types isn't just a matter of adding something to the enumeration in the digitalinputs code. Those numbers mean something to VRM as well, it will require updates to the entire pipeline every time.

Perhaps we simply need a "Generic input" type. Combined with a custom name, that should suffice as a catch-all?

nmbath, 2021-02-02:
@izak 

Understood about the upstream feed into VRM. It is what triggered this request. 

A generic input type with custom name, would sort of work. It would also need the ability to define what the logic state of the input is as well. The logic state can be translated to On and Off, low and high, open and closed, etc. This data is also transposed through to VRM. 

So if you extend the thought so you an define a digital input as Generic, Custom Name and Logic translation that would give flexibility.

Would you then allow VRM to display the custom name within the input type and selected logic state within input type. Without this the VRM would display the digital input as Generic and some Generic input descriptor. Thus not being very friendly.

"Generic Input" within VRM would not be very prescriptive if you had four of them.

mpvader, 2021-08-17:
I think the easiest thing is to add your items, @nmbath and @osaether. To make sure, could you explain what your additions are meant for exactly?

closed 2022-12-07 by izak