victronenergy/venus #749: Wifi intermittent Ver 2.60

Yuphorya, 2020-12-02:
Just something I've noticed with 2 different Pi's. When I put version 2.60 (Current latest) on a Pi 3B+ I would sometimes not be able to connect via Wifi, but over ethernet would be ok. Just for a test, I tried ver 2.6~42 and this seemed to solve the issue. Th R.Pi would be recognized on the router but could not connect via web or SSH. Is there a setting I should be looking at?

Yuphorya, 2020-12-05:
If anyone is reading this. I have been testing out the latest 2.60 and noticed that if I plug it into ethernet and get access to the console, then unplug and reboot. It is fine. Its not until subsequent reboots on wifi only that there is a problem. Very strange.

jhofstee, 2022-05-21:
A bit late, but rpi's is a community effort. Reporting issues here doesn't help (and it 1.5 year later by now)

closed 2022-05-21 by jhofstee