victronenergy/venus #751: Adding MbSwitch widgets to the Main Page

nmbath, 2020-12-02:
I have a need to add some MbSwitch widgets to the main page. The reason for this is to present the ability on the top menu page to switch a manual relay on and off. I am not sure if you are working on adding this yourself to the code base.

However I found that due to this line of code:

 if (model.children[i].description.localeCompare(service.description) > 0) 

Within the addService function in PageMain.qml, it was not possible. The reason is that the MbSwitch widget uses a property of name and not description. 

By adding this line of code to the MbSwitch.qml file I was able to add the MbSwitch widgets to the main page

property alias description: name.text

Example of what I have achieved:
[image]

There maybe a way already, which I have not found or you may consider this to be a hack or work around that you do not want in the system and for that I do understand. However, I would ask that you consider this change or something similar to allow MbSwitch widgets to be added to the main page as it is extremely useful for relay outputs and possibly other switch type functions, rather hiding them away in menus.

Many Thanks

jhofstee, 2020-12-03:
Hi, for the status quo just add the MbSwitch to the VisualItemModel there, but also increment the - 2 to to -3 in
``` for (i = 0; i < model.count - 2; i++ ) ```. That should avoid the .description from being compared.

closed 2020-12-03 by jhofstee

nmbath, 2020-12-03:
@jhofstee Thank you. Thats what I did initially, but actually wanted to move the item to the top of the list as well as be a little more dynamic. Shame that the change cant be made.

jhofstee, 2020-12-03:
I can be made, but the older will be dealt with first, see e.g. https://github.com/victronenergy/venus/issues/649. That affects all users..

nmbath, 2020-12-03:
@jhofstee, I wanted to ensure some items always appeared at the top of the list, before the sorted list, as in my view they are important and constantly scrolling down the list was not fun. Those items also happened to be MbSwitch widgets that did not have a description.

Thank you, I guess I will have to keep a locally modified version of MbSwitch.qml with the alias line in.

jhofstee, 2020-12-03:
no just, adjust the for loop, to step over it.

nmbath, 2020-12-03:
My mistake. I did step over them, however had been using the description to do that. I have changed my step over loop to use a different attribute and all working as it should. Sorry for the inconvenience.