victronenergy/venus #754: Inconsistency in settings for Relays

nmbath, 2020-12-03:
Within com.victronenergy.settings relay settings are not consistent.....

The initial state is broken out by relay, but the Function and Polarity are not:

Settings/Relay/0/InitialState
Settings/Relay/1/InitialState
Settings/Relay/Function
Settings/Relay/Polarity       

Really need these to be

Settings/Relay/#/Function
Settings/Relay/#/Polarity      

I have addressed this locally by modifying /data/conf and rebooting

  <Relay>
    <_1>
      <Function type="i" default="0" silent="False">2</Function>
      <Polarity type="i" default="0" silent="False">0</Polarity>
      <CustomName type="s" default="" silent="False"></CustomName>
      <InitialState type="i" min="0" max="1" default="0" silent="False">0</InitialState>
    </_1>
    <_0>
      <Function type="i" default="0" silent="False">2</Function>
      <Polarity type="i" default="0" silent="False">0</Polarity>
      <CustomName type="s" default="" silent="False"></CustomName>
      <InitialState type="i" min="0" max="1" default="0" silent="False">0</InitialState>
    </_0>
    <Function type="i" default="0" silent="False">0</Function>
    <Polarity type="i" default="0" silent="False">0</Polarity>
  </Relay>

Note the addition of "CustomName" as well so that you can assign a name to the relay. I have left the Original "Function" and "Polarity" in place as they are used within the QML pages, Generator start/stop and Tank Pump which would all need modification!

mpvader, 2020-12-03:
Hi @nmbath this will be fixed some day, when we add more functionality to the second relay than it has now (only on/off).

Which on the todo list, but not up too high I'm afraid; I expect it will still take a long time before we do that.

I'll close this issue now, as its not something that will be done soon.

closed 2020-12-03 by mpvader

nmbath, 2020-12-03:
I am happy to submit some diff files that will at least start to improve things.

mpvader, 2020-12-03:
Changing the paths is trivial, but making sure the settings are migrated properly on systems already out in the field are already less trivial & require proper testing. And, as long as we don't also go and invest time on making the second relay functional, not useful for any normal customer.

Conclusion: not something we're going to work on now, sorry.