victronenergy/venus #755: managed batteries: Log blocking status of MOSFETs to VRM

izak, 2020-12-08:
For managed batteries that support it, these paths have been available on dbus and the gui for some time now. Log also to VRM:
```
/System/NrOfModulesBlockingCharge
/System/NrOfModulesBlockingDischarge
/System/NrOfModulesOffline
/System/NrOfModulesOnline
```

izak, 2020-12-09:
Implementation of a warning path is in `can-bus-bms` branch `blocking-warnings`.

mpvader, 2021-01-21:
Changing this to v2.70.

With BYD now confirmed as having those fields not working properly (and us thus removing them from being visible), and also BYD being the only one that we're aware of that even has those fields; its all non-urgent anymore.

mpvader, 2021-06-28:
@izak whats the status of this?

all done and can be closed? and whats the status with BYD about this? Do you know?

izak, 2021-06-28:
Presently this is explicitly unsupported on BYD batteries, even though the Premium LV sends this information. That's because the information is incorrect in earlier firmware versions.

I tested a new firmware version where it works correctly. I do now know if that firmware is released yet. There were other issues with that firmware too. I will take it up with them and revert here.

mpvader, 2021-06-28:
ok but our side works right? And Pylon has official support for this, right? @izak 

if yes, please close this

izak, 2021-06-28:
Our side works correctly. Pylontech is in a similar boat to BYD: Wrapping up last minute changes. I'll go ahead and close it as this is being handled elsewhere.

closed 2021-06-28 by izak