victronenergy/venus #756: Incorrect paths in dbus_pump

nmbath, 2020-12-08:
Just like in #741  the dbus_pump is using /Management/ rather than /Mgmt/

com.victronenergy.pump.startstop0
ActiveTankService                                                      -
AvailableTankServices                                               {'notanksensor': No tank sensor}
Connected                                                                  1
DeviceInstance                                                          0
FirmwareVersion                                                        -
HardwareVersion                                                       -
Management/Connection                                         pump
Management/ProcessName                                     /opt/victronenergy/dbus-pump/dbus_pump.py
Management/ProcessVersion                                  0.3
ProductId                                                                   -
ProductName                                                            -
State                                                                          0

nmbath, 2020-12-09:
I have tracked this down to veilb_python and vedbus.py within the ext direct. 

While this is a very simple fix, I have attached a diff file with the changes in.

[vedbus.py..DIFF.txt](https://github.com/victronenergy/venus/files/5662747/vedbus.py.DIFF.txt)

izak, 2021-01-28:
This was fixed in 2018 already.

https://github.com/victronenergy/velib_python/commit/50b357c22b455bbb2798c25c61c2acba09e03058

We just need to update velib in the affected products. Potentially affected products (to be confirmed, determined by a simply grep):

- [ ] dbus-pump
- [ ] dbus-modem
- [ ] dbus-generator-starter
- [ ] dbus-bornay-windplus

closed 2021-02-05 by jhofstee