victronenergy/venus #758: Remote VE Configure error 61 on large systems 6+ Multi's

vejpasop, 2020-12-11:
Remote VE Configure <= 1.17 is broken on larger systems. (6+ Multi's)

Issue: Reading the the remote VE.Bus configuration exits with error 61 "Incompatible VE.Bus Configuration"

Cause:  The VE.Bus command to read the VE.Bus system configuration of a specific Multi sometimes gets the reply of an other Multi.
When requesting the configuration of all Multi's in order we start to receive replies of the first Multi's when once we address somewhere around the ~6th Multi.

I am still investigating the underlying reason.

mk2vsc V1.18 seems to fix the problem since that version adds an address filter to the reply of the command.

vejpasop, 2020-12-11:
We should consider back porting mk2vsc V1.18 to Venus V2.62.
I need to run some more tests before.

vejpasop, 2020-12-11:
This bug was introduced in Venus V2.50 (mk2vsc V1.16) and is caused by the increased MK3 baud rate (115200 vs 2400)
Affected are all MK3 based Venus devices. (MK2 CCGX still runs at 2400 baud)

Details:
mk2vsc needs to check if a product has been properly reset after changing VE.Bus configuration. It does this by comparing the vebus configuration of a Multi addressed by its short address through an MK2 command, to the configuration returned through a vebus frame with a full 32 bit address.  The MK2 command to retrieve the configuration triggers a broadcast on the VE.Bus, The MK3 filters the configuration frame with the requested short address and sends it to mk2vsc. Immediately thereafter mk2vcs requests the configuration by sending a vebus frame and forwarding all configuration frames to mk2vsc. Due to the increased communication speed in combination with a large number of Multi's some frames from the previous broadcast where forwarded and interpreted as the answer to the current requests.

mk2vsc V1.18 solves this issue.

mpvader, 2020-12-17:
Included in v2.70 and in v2.62~7

closed 2020-12-17 by mpvader