victronenergy/venus #759: Vebus: Shutdown on SOC deadlock

vejpasop, 2020-12-11:
Version info: The shutdown on SOC setting was introduced in Multi firmware version 451

Issue description:
In a system with a battery monitor the SOC is not synced to a Multi which has been shutdown due to a low SOC.

So to summarize:
1. System works ok. BMV SOC is synced to the Multi.
2. Batteries discharge to the lower SOC limit (for instance 15%)
3. Multi switches off.
4. SOC rises, but Multi doesn't know about it because we stop syncing the SOC when it is off.

Notes: 

- Resetting the Multi's does not bring the system back to live, since the internal SOC survives a reset
- SOC can still be written to a Multi which is shutdown in order to wake it up.
- Multi firmware does not have a provision yet to re sync the SOC on battery voltage.

izak, 2021-01-05:
A progress report. It appears not all Multis behave the same.

Further investigation is needed to find the firmware/hardware combinations affected by this.

Observed at the customer, who has 9 Quattros (2653), systemcalc would consider the Multi to be disconnected (because of [this routine][1]), and because there was no vebus service, it could not write the Soc across, leaving the Quattros off because of low battery Soc, and the system deadlocked.

When trying to reproduce on another system (Multi, 2600), the vebus service remained connected and the Soc is correctly written across, causing the Multi to switch back on.

At the very least the bug appears to be less pervasive than we thought.

[1]: https://github.com/victronenergy/dbus-systemcalc-py/blob/79e975513378a9aa0ea1074f0a45b7de54c35211/dbus_systemcalc.py#L817

vejpasop, 2021-01-07:
Additional info.
Although we did not check the state of the ```/Connected``` path we where able to write the SOC to the ```/Soc``` path to bring the system back. Multiple Multi resets where tried. by the customer and by ourselves.  It is unlikely that the ```/Connected``` state remained or ever was 0  after all resets. 

Izak verified with dbus-monitor that  ```/Soc``` was not written at the time. Also  ```/VebusService``` on ```com.victronenergy.system``` was blank (and therefore the reason for not writing the Soc).

The system was configured with a two wire BMS 
SOC low shutdown was set at 15% and SOC low restart at 16%.

I was not yet able to reproduce this issue.

jhofstee, 2021-06-04:
I am removing bug from this for the time being, since it seem to be stuck at "Further investigation is needed" / "I was not yet able to reproduce this issue".

mpvader, 2021-08-17:
Since impossible to reproduce, I'm closing this. If its an issue, it will come back.

closed 2021-08-17 by mpvader