victronenergy/venus #761: DVCC: Add configurable override charge voltage setting

izak, 2020-12-14:
In some cases, the installer may want to override the charge voltage sent by a managed battery.

One case is a temporary condition where there is an imbalance in a brand new battery that simply needs time to balance. A lower charge voltage limit can then be set, allowing room for the high cell and time to balance.

Proposed implementation:
* This will work similar to the present quirk architecture already in place for some batteries.
* As before, where such a voltage is set the lower of this voltage and the one sent by the battery will be used.

mpvader, 2021-01-21:
@izak how would enabling this new feature affect:

a) an ESS system with lead batteries, a Multi and solarchargers?
b) a non-ESS system of the same config?
c) a system with DVCC off? (I would assume no effect)

mpvader, 2021-01-21:
And @izak the title of this issue talks about a configurable override charge voltage setting. Is that what it is? Or is it a user configurable upper limit for the charge voltage?

izak, 2021-01-21:
> a) an ESS system with lead batteries, a Multi and solarchargers?

Only affects systems with a BMS, so no change in systems with a lead acid battery. The charge voltage configured on the Multi is used.

> b) a non-ESS system of the same config?

In a system with a managed battery, the override voltage will be sent to the solar chargers. In a system without a managed battery, this setting has no effect and the charge voltage configured on the Multi is used.

> c) a system with DVCC off? (I would assume no effect)

Completely ignored, only used with DVCC.

The labels in the GUI is "Limit managed battery charge voltage", and it is only shown if DVCC is on. So I believe this should be clear to the user.

> title of this issue talks about a configurable override charge voltage setting. Is that what it is? Or is it a user configurable upper limit for the charge voltage?

It is the second one, so technically the title of the issue isn't accurate. It is a user configurable limit, but it will not allow the user to exceed the voltage sent by the managed battery.

closed 2021-02-24 by jhofstee