victronenergy/venus #764: Cerbo GX requires resetting network settings to connect to a new WiFi network

IsaiahJTurner, 2021-01-06:
[bug]
Steps to reproduce:
1. Connect to a Wi-Fi network ("Network A") in Cerbo GX Wi-Fi settings
2. Turn off the router/move out of range from Network A.
3. Turn on a new router/move into range of a different Wi-Fi network ("Network B").
4. Attempt to connect to Network B from Wi-Fi settings of Cerbo GX.
5. Cerbo GX displays WiFi connection indicator in status bar for several seconds before disappearing and will not connect, IP Configuration will show Unknown, and password will not be remembered.

Expected behavior: Network B will connect.

mpvader, 2021-01-06:
Hi @IsaiahJTurner , thank you for the report. What Venus OS version were you using?

IsaiahJTurner, 2021-01-06:
2.6

It's happy to switch from Network A to Network B while Network A is still online but it will not switch to the unknown, never joined before, Network B after Network A is hard disconnected by unplugging the router. Reboot also does not help.

IsaiahJTurner, 2021-01-06:
Further info: just "forgot" Network B and then unplugged Network A. Goes into "Disconnecting" and then into "Connecting" I then went to General and "Reboot" since i figured that's what it would automatically do because of the reboot if no network connectivity setting and I had a theory that rebooting funks up the relatively recent network changes so venus will "never stop retrying" on wifi. I then went back to settings and connected to Network B and it worked fine. So as for reproducing that way, no luck. Not sure if it's because I tried very soon after disconnecting Network A or because I had previously connected to that network and used "Forgot" (DHCP lease would still be active + maybe need to reset network first)

I will take a panel off to get to the reset button on Cerbo tomorrow if I have time since I know it's pretty much impossible to debug weird network bugs without reliable reproduction instructions. This is definitely a reoccurring issue though, I keep needing to reset network settings on the device.

closed 2021-01-18 by jhofstee

mpvader, 2021-01-30:
Hey @IsaiahJTurner , thank you again for reporting this. The fix is now included in v2.62~14.

best, Matthijs